betaflight/betaflight #2889: Bf3.1.7 quad flip in Acro over motor 1

robschii, 2017-04-15:
Hi guys, 
I had some strange issue. My quad flips at full throttle over motor 1.  I save a log and I can see how motor output send 100 Prozent to the motor but motor1 has no power and then the quad flip. This is in Acro mode.  In Airmode it is fine.  I set idle speed to 8 because Boris tell me it can be desync. I fly dshot600. This setting I have to test tomorrow.   I have change motor and esc and I test a new Fc without any success.

chime13, 2017-04-16:
Greetings,
https://github.com/betaflight/betaflight/issues/2852
I'd try flashing it with 3.1.6 first.

robschii, 2017-04-16:
I had this error never before. I was flying 3.1.5 for a long time.   Then I used 3.1.6 and problems started. After 2 minutes of flying motor one make some noise. It sounds really confusing and the motor has no power. Quad is going deeper at motor 1.   Then I replace motor and esc  but the problem was still there.  Then I flash other BlHeli and bf3. 1.7  and now the copter flips at full throttle. Do I flash an older version the motor 1  has no power at full throttle but do not flip. I go back till 3.1.3  but the problem is still there. Some month ago I fly 3.1.3 without any problems. This is  really strange.   I have change the FC switch motors and Esc but it is every time motor 1.  Could some one explain it to me..  Sorry for my bad englisch

robschii, 2017-04-16:
Should I erase the FC by flashing?

robschii, 2017-04-16:
At #2817 are 2log s

robschii, 2017-04-16:
I have to correct myself in Airmode I have the same problem but after one minute.   I throw the copter in the corner.  Damn shit

robschii, 2017-04-16:
I test 3.1.2   and in works so far after full erase chip at flashing. I would test it again with full erase at flashing 3.1.7

robschii, 2017-04-16:
My setup  spf3  racerstar 30a v2 mit dys se 2008 2550kv

chime13, 2017-04-16:
Yes - Full erase after a full CLI backup of your configuration. Reinstall a "known good" firmware version and restore your settings in CLI - type "save" to store settings

robschii, 2017-04-16:
Maybe my fault?

chime13, 2017-04-16:
No @robschii , I do not believe it is your fault. There have been a high number of issues with the 3.1.7 release. I had to revert to 3.1.6 to get all my 3 quads back to flying and not flipping.

Eganwp, 2017-04-16:
> I have tested all versions since 3.1.2 but 3.1.2 is the first one without issue but I use full erease at 3.1.2 for the first time

There seems to be quite a number of people reporting this issue in various places, but I'm still not sure what's causing it as it is very similar to a standard desync. Can you please do us a favour and swap the desyncing motor 1 and ESC 1 over to another arm's position and report back of the issue follows the motor/ESC? If so we'll know if it's a possible FW issue or not as I'm not able to reproduce the issue personally.

Eganwp, 2017-04-16:
Also, which esc's and what version of blheli FW are you running on them?

19Gerhard85, 2017-04-16:
I have a similar problem, on punch outs the quad want's to flip.

FC: X-Racer F303 V3
Motors: Racerstar BR2205S 2600KV
ESCs: Foxeer F25A w/ latest BlHeli_S

Eganwp, 2017-04-16:
@robschii @18Gerhard85 What ESC communication protocol are you guys both using? DShot 600, 300, 150, Multishot or Oneshot?

robschii, 2017-04-16:
dshot 600    racerstar 30A v3 with blheli_s 16.65    i tried also 16.5    same problem    and i have swap the motor1 with esc with motor3 and esc      but it is everytime the the right backward motor     and i have tested it with multishot     same problem     i have to go back to version 3.1.2   with full erease chip than it is everything fine.      yesterday  i flash 3.1.3  without full erease chip coming from 3.1.7   and the problem was also there

19Gerhard85, 2017-04-17:
@Eganwp
I'm using Multishot on this build.

robschii, 2017-04-17:
I think it's not a esc protocol problem.  Its something in betaflight.   I got the problem, then I land the quad and disarm. And arm again and fly and it is away for a minute.

njenwei, 2017-04-17:
I was having the same problems with my quad after I upgraded to 3.1.7 from 3.1.0. It would spin out of control while flying. One motor (usually motor 2 or 3) would randomly seem to stop, although the blackbox indicates that it goes full throttle, and is trying to compensate for the fall. I have swapped the motor-esc positions around and it still happens to different motors. IRL I can feel that the falling spin is too slow for that particular motor to be maxed out. 

Also, when hitting full throttle, the quad would start to vibrate - the motors would oscillate between full and mid power. At 8k PID loops, when I plug in the battery, without arming, the motors would spin up by themselves, with the same vibrating motions as when flying full throttle 4k. Perhaps this is a problem with the motor output from the FC? It seems to exacerbated by the higher loop rates.

I have been flying this setup for a few months without problems, using the same min throttle/min commands, so it is probably not a desync issue.

FC: AirbotF4
ESC: Kiss 30A (ONESHOT125)
Motors: Emax RS2205

[image]

https://www.dropbox.com/s/doix1mdlgihh70x/BLACKBOX_LOG_20170417_111427.BFL?dl=0

https://www.dropbox.com/s/n97r9apembg0kl7/BLACKBOX_LOG_20170414_153210.BFL?dl=0

joseserran, 2017-04-17:
I have had the same issues with my Quad since I installed 3.1.7, first issue was something related with motor 1, then the quad fall over making some flips nad rolls. Second time both motor 1 and motor 3 had an issue and the quad did a perfect back flip until it crash on the ground. 

FC: OmnibusF4 SD
ESC: racerstar lite_s bb2 30A
motor: racerstar br2205 2600kv

My other quad with the same configuration but different FC target and version (3.1.6) does not have any issues yet.

FC: AirbotF4 
ESC: racerstar lite_s bb2 30A
motor: DYS SE2205 2300kv

Aileron8, 2017-04-17:
I spent a better part of the weekend trying to recreate the issue on seven different quads all running BF3.1.7 and nada. No desyncs other than one time with my BFF3 FC when a pair of caps had broken off. I replaced the caps on the ESCs and all was well.

robschii, 2017-04-17:
My teammates also fly 3. 1.7 without any issue at spf3 4/4k  it's really strange could it be a problem with the new win 10 creators update that something goes wrong at flashing. I have no idea what it can be

chime13, 2017-04-17:
No idea about Windows - I use Betaflight 3.1.1 Configurator on a Mac and EZ-GUI on my Android phone.
As far as my issues - all gone since downgrading to 3.1.6. They're flying much better than I ever expected - nothing like a good tune...

19Gerhard85, 2017-04-18:
[DVR Footage](https://youtu.be/PXa_AQBqyb0)

ShikOfTheRa, 2017-04-20:
I have same issue on mine. 

airmode on = no issues. (but audibly it appears there is less power when running airmode).

Without airmode, on full throttle, motor 1 axis dips (or opposite is rising - not sure which!).
It doesn't feel or sound like desync issue. I also replaced motor and esc... I tried pwm and oneshot. With direct input on bench, motor / esc are capable of sustaining full speed - they not die under load 

The dip starts maybe a second or so after full throttle. It builds up stronger almost like I windup.

I go up high to carry out testing. Backing off throttle a bit or enabling airmode allows recovery. Its scary after long drop and apply sustained lot of throttle...

I recall similar with multiwii years ago. To resolve they dropped max output from RC throttle to allow headroom for pid to function correctly. Maybe if one is at max, others could be limited?

Indy6, 2017-04-21:
Almost same here, my new 1104 7000kv 88mm flips 180 degrees on the right side when I push full throttle on BF 3.1.7...
I'll try 3.1.5

AnGeL85DR, 2017-04-22:
Hi, Forgive my English! I also found your same problem in acro mode with high PIDs unable to hold the axis and very often it tilts to the left, but today I tested the airmode and all problems have disappeared, I did a DVR of Try :)
https://youtu.be/0jQfnGQPhQg

robschii, 2017-04-26:
is there something in progress? becoause i testeted every version till 2.8.1 but it is not posisble to fly my quad.  sometimes it is motor 1 and sometimes itr is motor 3  which has no power

nathantsoi, 2017-04-26:
did you guys try airmode? are you seeing the same behavior even with
airmode turned on?

robschii, 2017-04-26:
airmode   = the same 
now i have changed one esc  and go back to 3.1.2 and blheli 16.4  with multishot

joseserran, 2017-04-26:
I turned to 3.1.6 and no issues yet. I will text if I notice something weird.

robschii, 2017-04-26:
first flight with 3.1.2 and blheli 16.4  was good    i will test some times more

KrackJack1, 2017-05-12:
hi friends.I have the same problem which u guys are facing..my setup omnibus f4 30av2 after updating beta flight 3.1.7 and BL Heli 16.6.....problem heading goes automatic on 180 deg..  fc mount.  after take off motor 2 shut down ..  i have checkd with new esc and new board omnibus f4...still stuck..

Ozzy33, 2017-05-15:
I have 1 quad with 3.1.7 (other 3 are 3.1.6) and 3 days testing, 162 packs w/o issues. Revolt, FireFly 22A, RS2206-2400 V3, 5050x3, 4S-1300, los freestyle.

jumatic, 2017-05-25:
So was using 3.1.7 with the Torrent which have SPF3 Firmware. 4/4 Khz 8000PWM. Airmode, Anti Gravity enabled. All the way it was flying fine. Changed to 16000PWM and motor 1 freaks out and quad spins over diagonal axis continously and crash. So changed back to 8000PWM all fine. Hope that helps in finding solutions.

closed 2018-04-21 by DieHertz