betaflight/betaflight #2890: BF 3.1.6 - NAZE - blheli passthrough may softbrick ESC

Zappelphilipp, 2017-04-16:
tried to flash my RS35a ESC's and 3 out of 4 had a corrupted config/firmware afterwards. Got all except one working again by reflashing it several times. One ESC althought was completly corrupted and just said **FLASH*Failed**. reflashed through BF 3.0.1 and worked perfectly. https://www.reddit.com/r/Multicopter/comments/65oojn/did_i_brick_my_rs35a_escs/

DieHertz, 2017-04-16:
`**FLASH*FAILED**` is not a big deal, it may happen even with perfect code. Without logs, it's hard to determine what has gone wrong.

Dingl3berry, 2017-04-16:
I'm New to this forum!.. 
but I believe he's referring to the Betaflight pass through issue we are all having!... 
I have a Brand new Betaflight F3 Flight Controller with brand new DYS XSD ESC's 30A, I was using the pass through function to update the ESC's, BUT after the 2nd one failed I canceled the update, only to find out those 2 ESC are now bricked, no longer are visible to Blheli... 

Now I'm trying to find a fix for my ESC's so far I've found the arduino nano C2 4-wire method in another forum, doing a quick check to see if there has been a fix or a simple update procedure before Desoldering me ESC's.

Eganwp, 2017-04-16:
Had this happen about 6 weeks ago on 3.1.5 and bought an Arduino UNO to reflash all 4 successfully the first try.

Matthias7Song, 2017-04-17:
Had same issue with X-racer f303 and DYS XSD 30a on BF 3.1.5 and two esc's were bricked. I switched to Fury F3 but have not yet recovered those bricked ESC's. I'll give it a try with BF 3.0.1 next time.

Dingl3berry, 2017-04-17:
Reading others comments! Makes me feel better! 
For a minute I thought it was something I did... 
thank you to all for reassuring me it wasn't me!... lol... 
I will try the Arduino Nano C2 method & I'll repost my results!...  Thank you!

Zappelphilipp, 2017-04-17:
@DieHertz made a ticket regarding the bug: https://github.com/blheli-configurator/blheli-configurator/issues/49

Matthias7Song, 2017-04-18:
Same issue? https://github.com/betaflight/betaflight/issues/2897#issuecomment-294967342

spuder, 2017-05-02:
Possible duplicate of #2319

closed 2018-04-21 by DieHertz