betaflight/betaflight #2891: Erratic PWM signal with RX but normal using the BF configurator Motors tab

sadigov, 2017-04-16:
BF version: Betaflight / BLUEJAYF4 3.1.7 Apr  3 2017 / 21:54:04 (e1c4b5c)
ESC:           FX20A V2 – 4 in 1 20A – Cicada – BusyBee2 – BLHELI_S 
FC:              Micro LUX F4

Hi All,

I built a mini-owl micro brushless, but can't get it to hover. I have correct motor directions and numbers, as checked using the Motors tab in the BF configurator. The motors have an erratic PWM input when using a radio TX, but are perfectly in sync when using the PWM signal from BF configurator's "Motors" tab. 
I started with the stock PIDs and it was terrible, PID tuning somewhat helped, but the jumpy PWM issue is still there. It seems like it is not possible to tune the PIDs. I suspect there is some kind of a software glitch.

What could be the problem? 

Cheers,
Shamil.

Juan-ma, 2017-04-18:
Im new in Betaflight but I will try to help. Maybe you have a problem in the radio TX? Have you try the Receiver tab in the Betaflight configutator? Do you see the proper values in the correct channels when you move the sticks? What trasnmitter/receiver are you using?

sadigov, 2017-04-18:
Hi Juan-ma,

The receiver-transmitter are solid, I have 6 other models I use with the TX
(Frsky Horus with OpenTX) and the RX (Frsky XM+). The ranges in BF look
good, same as with my other models.
 I soft-mounted the flight controller and most of the motor jitter issue
has been solved. At least the quad hovers, but still unstable. I think the
gyro is getting a lot of noise, will try to play with filters to kill the
high frequency oscillations.

Let me know if you have any suggestions.

Cheers,
Shamil.

useDforce, 2017-04-22:
I am having a similar issue at first i thought it might have been my bad soldering but after redoing most of the joints and its still occurring it must be something else. Similar to your situation when I fly it is present causing the quad to have like a dipping effect during flight like the motors are losing power for a tick. It is very inconsistent does not always happen at the same points or anything. When in betaflight I cant recreate the situation but as soon as i use my radio it occurs.

mikeller, 2017-04-28:
@sadigov: I am aware that this is just a crutch, but have you tried setting `gyro_lpf = 188hz`? Gyro noise can be quite harsh on micro frames (and since the motors tab in the configurator directly controls the motor output, you won't be noticing noise there).