betaflight/betaflight #2893: The barometer is not working properly

liguo-git, 2017-04-17:
F3  Turn on the barometer, set the height fail

nathantsoi, 2017-04-25:
hi @liguo-git, did you figure this out? if not, could you tell us what flight controller, what firmware version and possibly provide a configuration dump and photo of your setup?

liguo-git, 2017-04-25:
SP Racing F3
firmware  version 3.1.7 
inav normal

nathantsoi, 2017-04-25:
im confused, are you running the inav firmware? in which case, it's probably better to open up a ticket over here: https://github.com/iNavFlight/inav/

liguo-git, 2017-04-25:
hehe

nathantsoi, 2017-04-26:
haha, is that a yes? i'll close this, but reopen if this is a firmware bug in betaflight

closed 2017-04-26 by nathantsoi

mikeller, 2017-04-26:
Just a little addendum, in case you _are_ using Betaflight 3.1.7: Altitude display / altitude hold only work in 3.1.7 and earlier if BARO ***and** ACC are enabled.

jumatic, 2017-07-10:
Also have this issue with 2 boards. Omnibus F4 Pro V2 and the Nano F4 V2. It drifts down to about -4. Inflight i have after some seconds of flight +4m near the ground i started at.
[image]
[image]
It is connected to usb only. now for some hours. Could it be a heat isssue?
Tested with BF 3.1.7 and also 3.2 with/without acc. Same result.

mhusiak, 2017-10-08:
I'm having the same issue as jumatic. I can be less than 1m off the ground and the barometer reading could say something like 19m. Then -9m. 

Any solution for this?

mikeller, 2017-10-08:
@mhusiak: The barometer chips used on most flight controllers are from the cheap end of the spectrum, and often do not work well in an environment with lots of motion and moving parts creating air streams. One thing that can be done to try and improve the barometer readings is to shield the barometer from airflow with a piece of sponge or foam over it.