betaflight/betaflight #2897: Esc's/Motors not working after 3.1.7 update

fletch3d, 2017-04-18:
Have an mx nano oversky 32 FC board with brushless motors for my Scorpion 120.  Has been running 3.0.X without issues.  Esc's are BLheli flashed and the passthru has worked flawlessly.  After flashing several Kiss FC's and even my beebrain with 3.1.7 with great success, I had no such luck with my oversky naze 32 fc board.

Flashing of the 3.1.7 went well, setup modes, radio, pids etc.  However my esc's will not arm and motors will not start.  I tried to run blheli suite to get into the esc's, but it will not see the esc's at all.  When powered up the esc's beep once and that is it.  Not sure if anyone else has run into this?  I am not sure what to look for next?  Help!!

RipperDrone, 2017-04-18:
if ESCs just beep once it seems that either upper or lower limit of calibration (PPM_Min(Max)_Throttle in BLHeli) are set inconsistently. Try to manually set min/max to 1020 and 1980, respectively in BLHeli and write those values back to all ESCs. Maybe this fixes your single beep issue...

fletch3d, 2017-04-18:
Thank you RipperDrone.  That was my thought, however I cannot get to the esc's using the passthru from the FC.  BLHeli will not see the esc's at all.

RipperDrone, 2017-04-18:
If bootloader is bricked, you have to recover via FTDI cable, I guess...

fletch3d, 2017-04-18:
Ya.. Boot loader is fine now.  Passthru is the issue, I Think?  May have bricked my esc's..

Eganwp, 2017-04-18:
Yes. I had the exact same issues with my Hobbywing 30a blheli_s ESCs not showing in blheli_configurator back in 3.1.5. Occasionally all would show up, on a refresh only some, etc. Flashing them ended up bricking 2 of them. I restored the boot loader after buying an Arduino UNO and they have worked fine ever since. It's either an issue with pass through or with blheli_configurator. 

- Travis

fletch3d, 2017-04-18:
Thank you.  Where did you get the UNO?

Eganwp, 2017-04-18:
Haha one of the only things I've bought locally in store in a long long time! http://www.active123.com/

A $3 Arduino Nano also works though. I grabbed on from myrcmart for another dedicated project. 

Travis Fletcher

fletch3d, 2017-04-18:
Awesome..  Thank you..

Chris Fletcher

Matthias7Song, 2017-04-18:
May be related. https://github.com/betaflight/betaflight/issues/2890

McGiverGim, 2017-04-19:
Are you using the latest version of blhelisuite? I has been updated for version BF3.1.7 and CF2.0. If you have an older version it will not work.

fletch3d, 2017-04-19:
Thank you.  I did get the latest version of BLHeli, but ended up with the same issue.  Looking more like my esc's have been bricked.

McGiverGim, 2017-04-19:
Here you have the info
https://github.com/cleanflight/cleanflight/issues/2725
Take a look at the version

McGiverGim, 2017-04-20:
It seems that your link is the actual version, updated the April 1st. If you are using this version, the problem is other.

fletch3d, 2017-04-24:
Need some direction on how to use the Nano to reflash the bootloader on the esc's?  I tried to reflash the ESC using the Nano and it still doesn't read it.  I probably need to put the bootloader back on it, but not sure how to go about it.

nathantsoi, 2017-04-25:
does this help @fletch3d https://nathan.vertile.com/blog/2015/12/06/blheli-esc-identification/ ?

can we close this since it appears to not be an issue with the flight controller firmware?

closed 2018-04-21 by DieHertz