betaflight/betaflight #2899: SPRF3 - 3.1.6 -> 3.1.7 no arming

RipperDrone, 2017-04-18:
Currently doing extensive update of all my (13) boards to 3.1.7 for testing:

Moving SPRF3 target from 3.1.6 to 3.1.7 causes no arming (gyro calibrated, double-beep sounds ok, no other signals perceived). 

Reverting back to 3.1.6 and arming is ok again...

Anybody to replicate this behavior?

Arakon, 2017-04-18:
Nope, works fine here. Make sure you test without restoring a dump first, to see if it's a setting you are copying over. Also check your CPU load.

RipperDrone, 2017-04-18:
CPU load is @ 9%. I tried "fullchip erase", then it worked. Might have been dump overwrite, however I went through CLI diff and did not find any inconsistencies...

closed 2017-04-18 by RipperDrone

RipperDrone, 2017-04-18:
Found it - due to addition of antigravity switch, it seems the dump restore had wrong enum, therefore messing up the switch settings... had to re-configure. It's a pity that those switches don't have identifiers, so they are more robust against changes...

borisbstyle, 2017-04-18:
Switches do have unique identifier. It was the cli not displaying it that way.

RipperDrone, 2017-04-18:
@borisbstyle K, thanks for clarifying. How do you generally migrate your settings from one version to the next - save a diff and restore into new firmware via CLI is not robust enough?

MJ666, 2017-04-18:
Using the diff is what I always do and never had issues. Using a full dump between versions is not a good idea.