betaflight/betaflight #2901: 3.1.7 Failsafe full throttle in 3D mode

Zappelphilipp, 2017-04-18:
using the default 3D mode failsafe works as expected. We are talking about the default failsafe settings in Stage 2, 1000ms guard time and then drop.

But if i am in DISABLE_3D_KILLSWITCH mode, the motors spin up to pretty high revs (not exactly sure if full throttle) and then stop. Of cource failsafe_throttle is set to 1460 which is my mid throttle. I also tried to change the failsafe mode to LAND and set the landing revs to 1460, same result. interesting is, that the motors spin at high revs for the duration which the guard time is set to, then suddenly slow down to idle revs for about a second and then stop. If i set guard time to 0, the motors rev up very high for a splitsecond, then slow down to idle for a second and then stop.

RipperDrone, 2017-04-20:
Good point - let's see: If you fly 3D enabled, thorttle range is fwd from 1500-2000 and reverse from 1000-1500. When disabling 3D, the FC output range to the ESCs is 1500-2000, but RC throttle range is remapped by FC so that full min to max stick can be used. 

At least this was my understanding so far.

Consequently, failsafe throttle (stick input) has to be 1500+x in 3D mode, but 1000+x in 3D disabled mode. This does not seem to be the case in the current firmware...

Zappelphilipp, 2017-04-20:
Yep, should be correct. Can you link me the line of code where you found the problem?

Zappelphilipp, 2017-04-20:
hmm, just found
` .failsafe_throttle = 1000,                       // default throttle off.`
in _failsafe.c_. Where does the remapping happen?

RipperDrone, 2017-04-20:
mixer.c handles 3D feature...

closed 2018-04-21 by DieHertz