betaflight/betaflight #2904: configurator connection problems on linux

mhv-shared, 2017-04-19:
I was having trouble connecting through the configurator on linux (Ubuntu 16.04) - serial port would be opened successfully but connection would time out.

I noticed these lines in syslog:

    Apr 19 08:03:50 mh-EX58-UD5 mtp-probe: checking bus 4, device 16: "/sys/devices/pci0000:00/0000:00:1a.1/usb4/4-1/4-1.1"
    Apr 19 08:03:50 mh-EX58-UD5 mtp-probe: bus: 4, device: 16 was not an MTP device

Purging libmtp-common fixed the issue but is a bit drastic. Possibly it could be avoided with a udev rule, or maybe a change in the firmware could make it play better with mtp-probe?

cs8425, 2017-04-19:
try uninstall `modemmanager` package.
`sudo apt-get --purge remove modemmanager`
This package seems to send something to serial port cause error.

Hambrook, 2017-04-23:
Have you tried adding yourself to the "dialout" user group?

nathantsoi, 2017-04-25:
did that work @mhv-shared? can you close if it's ok?

mhv-shared, 2017-04-25:
Closing as I am unable to reproduce the specific problem to test these different fixes in isolation.

closed 2017-04-25 by mhv-shared