betaflight/betaflight #2906: OSD overlay video image.

goseeped, 2017-04-19:
I just build a new brand betaflight f3 quad, I was at home and test everything (so far so go) then I went out and start flight in the middle on my first flight the OSD start overlaying the video from the camera with a gray background not allowing to see anything, the craft wasn't too high and the crash wasn't that bad.

When this start happening looks like this image then you see all gray background.

[image]

This happens to me using: betaflight_3.1.7_BETAFLIGHTF3.hex didn't have time to test yet with the previous version 3.1.6.

if you need more information let me know.

Hardware:
-VTX: tbs unify pro V2 using smartaudio (5 Volts) - UART3
-Cam: runcam Rotor Riot edition (5-17 Volts)
-Flight Controller: betaflight f3 with a (4S vbat)
-Receiver: X4R-SB (smartport) - UART1

Note: Camera and VTX share the same ground.

thanks,

Arakon, 2017-04-20:
The gray background is what you'd get when there's no video signal. Are you sure it's not the camera/wiring failing?

goseeped, 2017-04-20:
This is a new brand quad everything is new in it including cables but I guess I could make voltage and connectivity tests today and make an update here.

I wire Ram to 5V since my VTX take just 5V and I used the 5V pin from the softserial UART2 since I'm using the sbus and the RAM GND is wired with the CAMERA and VTX so both of them share the same GND.

There is something wrong with that wiring?

When I was flight 90% of the screen was full gray, I can only see the OSD information and 10% the camera, the image above were at home trying to reproduce the issue.

nathantsoi, 2017-04-20:
Do you have an LC filter on the camera power supply?

goseeped, 2017-04-20:
@nathantsoi thanks for your comment but Not at all.

> I wire/jum Ram to 5V since my VTX take just 5V then I used the 5V pin from the softserial UART2 (for the Rotor Riot camera) since I'm using the sbus the RAM GND is wired with the CAMERA and VTX so both of them share the same GND.

The 5v pins from the 'betaflight f3 FC' shouldn't be clean? I can get it if I had connected the camera to vbat.

goseeped, 2017-04-21:
Well, I did tests and here are the results.

1) I check connectivity, voltage and wiring (everything OK)
2) I check the Camera and VTX itselft. (everything OK)

I did not change anything or resolder anything but I just downgrade to betaflight 3.1.6 and my issue goes away, I try to search in the commit history changes related with OSD/battery cleanup in the latest version didn't find anything relevant although I'm not familiar with this code at all.

Thanks for the comments.

goseeped, 2017-04-23:
UPDATE: I thought the issue was gone so I did my homework and start digging into it, it is not a betaflight software issue, long story short (DO NOT USE 'betaflight f3 fc' RAM pin on 5V IT DOES A TERRIBLE JOB ON THE OTHERS 5V PINS OUTPUTS) it has hard time trying to deliver constantly 5v.

BTW a LC filter is not needed.

Since this is hardware related issue going to close this ticket.

Thanks,

closed 2017-04-23 by goseeped