betaflight/betaflight #2909: ICM-20689 SPI Frequency F4 target

flighttronics, 2017-04-20:
Data sheet say max 8MHz on SPI for ICM-20689
"Host interface: 8 MHz SPI or 400 kHz Fast Mode I2C"

Seams to run at 10.5MHz on F4
On F7 it is   SPI_CLOCK_STANDARD      = 16,  //06.57500 MHz  witch is ok...

Driver uses:
spiSetDivisor(ICM20689_SPI_INSTANCE, SPI_CLOCK_STANDARD);

bus_spi.h

#if defined(STM32F4)
    SPI_CLOCK_SLOW          = 128, //00.65625 MHz
    SPI_CLOCK_STANDARD      = 8,   **//10.50000 MHz**

kc10kevin, 2017-04-22:
Been running the FuryF4 (ICM-20689) for months now at 10.5MHz.   No issues at all.  

Setting the 20689 clock=16 on the F4 would bring it down to 5.25 MHz.  

If it ain't broke, don't fix it.  :-)

flighttronics, 2017-04-22:
Sorry, I think it should be fixed... 
I guess they have tested it to work with all chips on 8MHz.
How can you grantee it will work on all boards and with all sensor chips?
As I don't have the full picture of the implications of just changing the  SPI_CLOCK_STANDARD for all F4 targets.. I want a discussion around that or just  make a change in the gyro driver to set it to correct clock freq.

MJ666, 2017-04-23:
I think the GYRO/ACC is overclocked for many of the actual targets (I2C and SPI). May be a option to control the SPI speed per port will be fine but I would not turn it down this far since this speed has quite some influence on performance and the F4 has not the best factors for good adjustments. Till now I have not heard about instabilities related to the overclocking.

closed 2018-04-21 by DieHertz