betaflight/betaflight #2914: Betaflight doesn`t arm

MaxHalla, 2017-04-21:
Hi,

I have the Revolt 2 and flashed it with Betaflight 3.1.7 a month ago. It worked well. Now I reflashed the Revolt 2 with the same firmware, but it doesn`t arm. When I power it up with the battery and arm the switch, then there doesn`t come sounds. Only if I go with the powered Drone to the computer and connect it and disconnect it, then the quad is arming.

Can you helping me? 
And sorry for my bad english.

MaxHalla, 2017-04-21:
Update: the quad is arming with 3S Lipo but not with 4S Lipo.
Why?

RipperDrone, 2017-04-21:
Define upper voltage limit in BetaFlight Configurator to take 4S into account. If voltage is above the given threshold, it will not arm (same as if it lower than lower voltage limit)

MaxHalla, 2017-04-21:
Where can I setup 4s into account?

RipperDrone, 2017-04-21:
Oh, I was wring - sorry. Mixed it up with HV LiPos. I had to enhance the cell voltage max from where it was to 4.5V to take HV LiPos into account. Otherwise no arming. So your case seems different, my apologies...

RipperDrone, 2017-04-21:
Any LED signals from the FC when not arming? Usually the FC throws codes what keeps it from arming...

MaxHalla, 2017-04-21:
No, normally the FC is beeping while arming. Now there is no reaction from the FC...

wind0r, 2017-04-21:
sounds like the voltage rail isnt clean enough for that SSG. try raising moron threshold

MaxHalla, 2017-04-21:
I don't know, what you mean. My translation isn't the best. How can I raising moron thresold?

wind0r, 2017-04-21:
Use the cli and type "get moron_threshold" and the choose a higher number and write "set moron_threshold number" and save after that. Raise the number until your quad arms

RipperDrone, 2017-04-21:
e.g. type into CLI tab: 

set moron_threshold = 100
save

RipperDrone, 2017-04-21:
typically it's against noisy gyros which do not settle down calm enough before arming...

RipperDrone, 2017-04-21:
SSG = super scrappy gyro :-)

MaxHalla, 2017-04-21:
Thanks, it workes! I have set it to 100 and now it is  working very well. Thank You!

RipperDrone, 2017-04-21:
@wind0r - you hit it!

closed 2017-04-25 by nathantsoi

abdul1212, 2017-08-30:
nice get moron_threshold and set and save thanks bro

SRTDRE76, 2017-09-05:
Are these serious commands cause same problem all other switches encode and function as should the arm shows toggle move but does not engage.

jflyper, 2017-09-06:
@SRTDRE76 `moron_threshold` only changes the angle that is considered "arming-safe angle".

For further assistance, please visit the thread in RCG.
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread