betaflight/betaflight #2916: Reciever Tab Not Working In BetaFlight

ItsNovaFPV, 2017-04-22:
Hey guys so I just got my Eachine Wizard x220 with the SPRacing F3 Flight controller and I had an issue. The quad binds with my FS I6 but in the receiver tab of my beta flight there is no stick movements. I've looked at every video. I have everything set to PPM and it should work. Thanks for your help.

hydra, 2017-04-22:
use forums for support

nathantsoi, 2017-04-23:
@ItsNovaFPV, can you arm? e.g. do you know if the receiver is working at all or if the configurator tab is actually broken?

dleblanc67, 2017-08-17:
Any update?  I have Eachine Wizard X220 with Turnigy Evolution iA6C receiver.  I can bind the transmitter/receiver.  Betaflight sees the drone and I can see it rotate in Betaflight.  However, the Receiver tab when first clicked, shows colors, then they disappear.  No movement on the sticks causes any output to Betaflight.  I have SBUS configured on transmitter and controller for SBUS.  I have UART3 Serial Rx enabled.  Still nothing works.  Nor does the drone respond to the transmitter.

mikeller, 2017-08-17:
@dleblanc67: GitHub is for discussion and questions around development of Betaflight, not for support or configuration issues. Please have a look at (RCGroups)[https://www.rcgroups.com/forums/search.php].

closed 2017-08-17 by mikeller

NotmanyFPV, 2017-09-08:
In reciever tab set 1000 1500 2000 lol I just had the same thing on new GT200 xr4sbus :p

jkapdrone, 2017-09-12:
NotmanyFPV I tried this and this didnt work for me. I have the qx105 bat ,taranis plus frsky rx. No stick movements in betaflight. What could i be doing wrong??

NotmanyFPV, 2017-09-14:
Ossh, same thing happened to me with GT200 crusader and my taranis x7 a
week ago, I updated firmware to latest and it worked ae, despite it saying
on box not to update past 3.1.
works sweet Maiden flew it last weekend, dream.

Can you send a picture of your reciever tab?

jkapdrone, 2017-09-14:
NotmanyFPV thanks for your help. I got it going. An update probably would have did the trick. But my failsafe values were at 0. I read some information online about setting them to 885-2115 and that gave me colors and values in the receiver tab. 

Thanks for responding

NotmanyFPV, 2017-09-14:
Wicked! lol no problem, many a nights looking blankly at my quad beeping
trying to decipher online forums to apply to my situation haha this one
caught my eye as I was stuck on the same day with issue :)

Good times

rinyheuvelman, 2018-05-02:
It's the update indeed, same issue here. I also reconnected the receiver as ibus.
[SPRacingF3-Manual-latest.pdf](https://github.com/betaflight/betaflight/files/1966822/SPRacingF3-Manual-latest.pdf)
[image]