betaflight/betaflight #2918: Mid throttle oscillation bf 3.1.7 / bf 3.1.6

flighttronics, 2017-04-22:
Upgraded to 3.1.7 and got allot worse on the mid throttle oscillations.
It is not 100% same throttle but pid:s are the same.

BF 3.1.7
[image]

BF 3.1.6
[image]

RipperDrone, 2017-04-22:
filter defaults might have changed from 3.1.6 to 3.1.7. Have you tried to make sure filter settings / types are exactly the same (CLI get type to check filter types...)

flighttronics, 2017-04-22:
I took the dump from 3.1.7 into 3.1.6 so I should be the same

closed 2018-04-21 by DieHertz