betaflight/betaflight #2919: Spektrum SRXL: No Telemetry Data with Naze rev5

voodoo-bravo, 2017-04-22:
Hello, 

I recently purchased a SPM4649T Receiver to use it with my Naze32 Rev5 and Betaflight v3.1.7

What I did to set up the Receiver:

* Connected VCC to Bat-Sensor on Naze
* Connected 5v from the Naze directly to the Receiver
* Connected the Serial-Wire of the Receiver to the TX-Line of UART2 on the Naze
* Enabled 'VBAT' (shows correct Voltage in the Configurator-App)
* Enabled 'Serial RX' on UART2
* Set 'Receiver Mode' to 'Serial-Based receiver'
* Set 'Serial Receiver Provider' to 'SRXL'
* Enabled Telemetry
* Save and Reboot

The control-link works as expected and the channels are correctly displayed in the Receiver-Tab in the Configurator.
Problem is, that i don't get any voltage readings on my DX6G2.

Auto-Config on the Telemetry-Screen of the Transmitter reveals 4 Sensors: RPM; Temperature; RxV; Flight Log
RxV is constantly 0.0v, if I manually add a "Volts" Sensor, it also shows 0.0v

As a test, I hooked the Receiver up to the FuriousFPV PikoBLX on my micro-quad and did the above mentioned Steps again. 
After that, the Auto-Config on the TX found two additional Sensors (Volts and Powerbox) and the Battery-Voltage was correctly displayed on the Telementry-Screen of the Transmitter.
So I can rule out, that the Receiver is defective.

Here's a CLI-Dump: [bf317_naze_dump.txt](https://github.com/betaflight/betaflight/files/948900/bf317_naze_dump.txt)

Any Idea what causes this issue?

Greetings,
Markus

chriswhiteuk, 2017-07-06:
I'm experiencing the same issues but on a Betaflight F3 board with the same receiver, and same radio. Yet someone else on teh RC Groups forum has a DX9 with this receiver and a Betaflight F3 and it works flawlessly. I'm a bit stumped now. Have tried to re-flash and wipe settings and change UARTS with no success. Would be grateful if anyone has any ideas!

mikeller, 2017-07-06:
@chriswhiteuk: I do not have experience with Spektrum hardware at all. But presumably they are firmware upgradeable, and the discrepancies can potentially be put down to different firmware versions on the RX?

chriswhiteuk, 2017-07-07:
@mikeller They are indeed. My radio is running the latest firmware. I may try downgrading a version or two as a test. I''ve also tried reverting to Betaflight 3.1.6 which makes no difference. The RX itself isn't upgradeable as far as I am aware. 

I'm assuming I have things wired correctly hardware wise - the serial port on the RX is wired to UART1 TX. (As I understand the SBUS/DSMX port is RX only)

Setting SRXL seems to work, and all my channels work correctly as well as the reciever reporting it's built in RSSI back to the Radio, proving the Telemetry on the reciever works. 
It's almost as if the FC itself is not sending any data back out of the TX UART.

mikeller, 2017-07-09:
Have you enabled telemetry?

chriswhiteuk, 2017-07-10:
Yes, Telemetry enabled in the configuration tab. Correct Uart port enabled and no telemetry options selected on the ports tab (Which I assume is correct for Spektrum?)

mikeller, 2017-07-10:
Yes, that sounds all correct. Can you post a CLI `diff`, maybe there is a smoking gun in there.

chriswhiteuk, 2017-07-10:
No problem, I appreciate the help! It may be that the issue is with the receiver or the radio. Just trying to narrow things down. Although it's making me consider switching to FrSky even more...

```
# diff
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)

map TAER1234
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
aux 0 0 0 900 1300
aux 1 28 1 900 1300
aux 2 1 1 1300 1700
aux 3 13 0 1700 2100
set min_throttle = 1020
set serialrx_provider = SRXL
set acc_trim_roll = -2
profile 0

rateprofile 0
rateprofile 0

# 
```

mikeller, 2017-07-10:
No smoking gun there, sorry. :-(

chriswhiteuk, 2017-07-10:
No problem, thanks for taking a look!
In that case I can only assume the problem is with the receiver or the radio. I'll do some more digging from that side. Otherwise it's time for a Taranis!

Thanks again.

chriswhiteuk, 2017-08-03:
Potential fix
https://github.com/betaflight/betaflight/pull/3743

closed 2018-04-21 by DieHertz