betaflight/betaflight #2925: Increase one motor on random time

Otakee, 2017-04-22:
Hi everyone,

I used a setup with :

Flip 32 F4 : Airbot 3.1.6, 3.17 and another version
Little bee ESC Spring 35A : 16.6 Bheli
F40 Pro motor 2600kv
Default PID,
FS-I6 Radio

I have two problem, first i detected a noise on gyro yaw and after 1 or 2 minutes flight time my motor 3 increase throttle on 100% value and crash. The problem appear on ACCRO, HORIZON, ANGLE mode. Any idea ?

[BLACKBOX_LOG_20170422_203500.txt](https://github.com/betaflight/betaflight/files/949293/BLACKBOX_LOG_20170422_203500.txt)

HackIT, 2017-04-23:
Put capacitors(25v 220-470uf on each esc for a 4S setup or just put 25v 1f on the main power lead...) and softmount your beast.
For 5-6S you should use some 35v capacitors.

Otakee, 2017-04-23:
@HackIT It's due to littlebee esc ? I looking for a thread on rcgroups where people talk about it

HackIT, 2017-04-23:
@Otakee BLheli_S is done for multirotors with damped light enabled and no way to disable or smoothing it...
I'm the kind who likes the facts so I've found a demo about what damped light does on your hardware...
https://www.youtube.com/watch?v=ZhvuQJPGHzU

Also, here you can find some antivibration screws:
https://www.aliexpress.com/item/High-Quality-4-PCS-Realacc-M3-7-6-Flight-Controller-Fixed-Screws-for-FPV-Racer-RC/32800650354.html

After this you should get rid of all bad vibrations that makes your quad twitches...

Hope it helps.

Otakee, 2017-04-24:
@HackIT Thanks !
 I also added a antivibration on FC. I disabled temporarily DSHOT and problem is fix

HackIT, 2017-04-24:
@Otakee Just to avoid you making a stupid mistake.
If you do use capacitors, you have to be careful about one more thing...
NEVER short anymore neutral and phase... capacitors keeps electricity... and it takes a load to vanishes by itself.

Hope it helps.

closed 2018-04-21 by DieHertz