betaflight/betaflight #2928: PID einstellen, in welchem Mode

gerse70, 2017-04-23:
Hallo,
eine dumme Frage in welchen Mode soll ich die PID einlernen. 
Den Acro-Modus habe ich nicht!! Habe neuste Beta Versin am FC
Könnte Air Mode oder Haedfree anbieten.

Mit freundlichen Grüßen
Gerson Jaek

Noctaro, 2017-04-23:
Hallo,
eigentlich ist dies eine seite für Probleme, nicht für generelle Fragen, aber der Acro Mode ist immer dann aktiv, wenn weder Air, Angle oder sonst etwas aktiv ist.

Lg

gerse70, 2017-04-23:
Also wenn ich keine Mode hinterlege bin ich im Acro Mode, ist das richtig?

Noctaro, 2017-04-23:
Genau

gerse70, 2017-04-23:
was ist der TPA-Wert

gerse70, 2017-04-23:
was ist der TPA-Wert, wo stellt man ihn ein

sinned88, 2017-04-26:
Das hier ist kein Support Forum. Für solche Fragen bitte hier: http://www.rcgroups.com/forums/showthread.php?t=2464844 
Oder auf Facebook

closed 2018-04-21 by DieHertz