betaflight/betaflight #2935: Very strong oscillation, BUT ONLY on roll-axis?

ltwin8, 2017-04-24:
Hello Community,

I hope anyone can help me.... I build a little 130mm Copter based on the KingKong 130 GT frame.
The setup is:

DYS BE1104 7000KV
3020 Prop
Racerstar 4 in 1 20A V2 ESC (BB2) (DShot600)
Omnibus F3 flightcontrol with SD and OSD, but OSD not used
X6B Receiver (I.Bus)

The flightcontroller is soft mounted but tight because there was no other possibility and 180degrees upsite-down which I have configured and the movements from Copter match them in GUI.

Now my problem:
The copter hovers "Okay" , not the best flying....
But when I go for more throttle the Quad makes a noise what sounds something like a "Cricket" (tone goes high-low-high-low) but faster

And in Blackbox I saw that there is a strong oscillation, on roll most nearly the same on pitch but not so much on yaw.

andre92, 2017-05-04:
According to blackbox, you have a strong low-frequency oscillation. I suspect your PID gains are too high. Could you zero out your D gains, half your P and I gains and see if that removes the vibration?

ltwin8, 2017-05-04:
Hallo andre92, thanks for your answer, the PID is stock from BF, but I could try to make that changes

Thank you

closed 2018-04-21 by DieHertz