betaflight/betaflight #2936: BetaflightF3 RSSI on OSD reads 99, only.

Reinsettle, 2017-04-25:
On the Betaflight F3 FC the RSSI on the OSD reads 99 always. I heard that in 3.1.7 fixes the inversion but i am using 3.1.7 and it still isn't working. Everything else from telemetry works fine. Not sure if there is a way to manually invert the signal in CLI. Any advice?

jflyper, 2017-04-25:
@Reinsettle 
We need to know your receiver type, how you connected RSSI signal to BFF3, and CLI diff output.

Reinsettle, 2017-04-26:
It's an X4R-sb and smartport is connected to TX1 but I'm not sure what you mean by diff, so here is a dump. sry  # Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name Space One
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A00
resource MOTOR 8 A02
resource MOTOR 9 A08
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1
smix reset
[... 412 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 127
set rc_rate_yaw = 122
set rc_expo = 24
set rc_yaw_expo = 23
set thr_mid = 50
set thr_expo = 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1570

mikeller, 2017-04-27:
@Reinsettle: Just type `diff` instead of `dump`. ;-)

Reinsettle, 2017-04-27:
feature -BLACKBOX
feature -SDCARD
feature RSSI_ADC
serial 0 32 115200 57600 0 115200
aux 0 0 0 1700 2100
aux 1 28 1 900 1700
aux 2 4 2 900 1700
set fpv_mix_degrees = 29
set digital_idle_percent =  4.400
set motor_pwm_protocol = DSHOT600
set ibat_scale = 258
set gyro_notch1_hz = 0
set failsafe_delay = 15
set acc_hardware = NONE
set pid_process_denom = 1
set osd_units = METRIC
set osd_rssi_alarm = 46
set osd_cap_alarm = 1355
set osd_time_alarm = 9
set osd_vbat_pos = 2433
set osd_rssi_pos = 26
set osd_flytimer_pos = 0
set osd_ontimer_pos = 374
set osd_flymode_pos = 364
set osd_horizon_pos = 200
set osd_current_pos = 2456
set osd_mah_drawn_pos = 2445
set osd_power_pos = 1
set vcd_video_system = 2
profile 0

jflyper, 2017-04-27:
And how you connected your receiver's RSSI signal to the BFF3?

mikeller, 2017-04-27:
Looks like you have not enabled RSSI inversion. Try `set rssi_invert = on`.

Reinsettle, 2017-04-27:
Ok so i set rssi_invert = ON and it's just blinking 0 now. So it wasn't that, unfortunately

Reinsettle, 2017-04-27:
The signal is coming from the smartport of the x4r and going to TX1/UART1. In the ports tab of betaflight UART1 is set to smartport for telemetry output.

mikeller, 2017-04-27:
@Reinsettle: That's the telemetry connection, and the signal is sent from the flight controller to the X4R. For telemetry, you will need a modification like this: http://fpvlab.com/forums/showthread.php?46052-OSD-Core-PNP50-RSSI-with-receiver-X4R-sb-from-Frsky-Need-a-ground-or-not
And then you will need to connect the RSSI out on the X4R to the RSSI in on the BetaFlight F3.

Reinsettle, 2017-04-27:
I see... Not having it connected would probably be the problem 👍 Thank you

Reinsettle, 2017-04-27:
I don't know what more i can do. I have soldered a wire from the small pad on the x4r and on the top of the BFF3 there is a small pad with rssi above it that i soldered to but it still isn't working. I've tried both set rssi_invert = on /and = off and still not working.

jflyper, 2017-04-27:
Or, if you only want the RSSI to be displayed on the OSD, you can choose to use the transmitter side loopback method.

Reinsettle, 2017-04-27:
For anyone else with any RSSI problems, If you are using a Taranis and don't feel like soldering and figuring out how the rssi pad on the FC works just set it up like this. It is very accurate as well

http://djwoodz.com/2016/04/14/Betaflight-RSSI-on-MinimOSD-over-SBUS-X4R-SB/

pstub, 2017-06-13:
I'm having the exact same issue... RSSI is fed back on channel 8, set up in BetaFlight configurator, but the OSD still shows 99% the entire time. Haven't found a way to choose where the OSD should take the info from. In the past I've only used MWOSD, and then I could choose to have RSSI value taken from FC. Looking for a place to choose the source of the RSSI on the BF FC OSD...

jflyper, 2017-06-13:
@pstub If your RSSI is on AUX8, then your RSSI channel will be 12.

pstub, 2017-06-13:
Gotta double-check! It SHOULD be on AUX4/Channel 8. Will check when I get home and let you know!

pstub, 2017-06-13:
I double-checked now. On any other channel than Channel 8, I get a blinking 0% in the OSD. On Channel 8, I get a steady 99%. Problem is that when I was out flying on Sunday, I got a low RSSI warning from my Taranis, but the OSD still showed 99%... So that's where my problem is...

jflyper, 2017-06-13:
Change your RSSI channel to one of AETR (Channel 1~4) and check if stick movement is reflected to what RSSI display on configurator and OSD say.

pstub, 2017-06-13:
Yes, that actually worked! So it COULD actually be working great, but the problem is that the Taranis was giving a fake warning? Cause I wasn't that far out, really. Could just have been a glitch when switching from one receiver antenna to the other, without the OSD even having time to react?

jflyper, 2017-06-13:
My XSR occasionally complains loss of signal and telemetry on the bench, too.
Thinking about updating the firmware...

bigtuni, 2017-07-01:
Hello, I am using a Graupner MX-16 transmitter with a GR-12l SUMD (SBUS) receiver and Betaflight F§ as FC. I have no idea how to set a channel on the TX for the RSSI, or where to find it. I currently always see 99% on the OSD. Chaning to channel 6 in the BF-config for the RSSI changed this to 50%. 
Any ideas? Appreciate your support.

mikeller, 2017-07-01:
@bigtuni: GitHub is for discussions and questions around development of the firmware. For configuration questions, please have a look at RCGroups: https://www.rcgroups.com/forums/showthread.php?923419-RSSI-pins-for-various-receivers/page24

closed 2017-10-23 by jflyper