betaflight/betaflight #2937: External IIC bus unresponsive using multiple tested peripherals

avidtech, 2017-04-25:
Hey everyone,
First board with betaflight. first time posting in this repo.

First things first, I'm using a Flip32 Omnibus V3, this one:
[http://cdna3.zoeysite.com/Adzpo594RQGDpLcjBynL1z/cache=expiry:31536000/resize=fit:max,width:1200/compress/http://s3.amazonaws.com/zcom-media/sites/a0iE000000GWIo4IAH/media/catalog/product/1/0/10-4-2016_7-00-39_pm.jpg](url)

I am using a 3 wire SPEK2048 protocol Rx, no SBUS or PPM ins (unless the spektrum is one of the two or shares with ethier of those busses, no idea), and a UBLOX GPS on UART6. I also notice that the CLI prints UART0/2/5 instead of 1/3/6, but I believe that's likely arbitrary.

I have tried hooking up a few different things, BMP085, BMP180, HMC5883l (The gy-271 and 273 variants, even though the 273 has a known issue req. pullups if I am correct), haven't had any luck with the configurators settings, setting the hardware through the CLI, etc. I am used to compiling with arduino and flashing not using this GUI stuff. It's very cool, I'm just not used to it.

Did I miss something? Is there a specific command set to define UART3 as I2C? Would anyone know if there is a hardware change necessary (like removing a jumper)?

Is this related  #1582? 

Thank you in advance, looking forward to flying with betaflight!

nathantsoi, 2017-04-25:
hey @avidtech, thanks for your feedback. I'm stoked you're flying betaflight! now, what exactly is going on?

- if you're using the spektrum port your receiver needs to be set to serial rx in the configurator on uart3, sounds like you've got the right protocol configured

- as for the i2c, @jflyper has a great guide here: https://www.rcgroups.com/forums/showthread.php?2792346-Exotic-OMNIBUS (search for Configurable I2C for OMNIBUS F3 AIO and F3 Pro)

closing since this doesnt appear to be a firmware bug. if that doesn't answers your question, feel free to read/ask [here](https://nathan.vertile.com/blog/2016/07/07/omnibus-typhoon-miniquad/#flight-controller-setup), [here](https://www.rcgroups.com/forums/showthread.php?2792346-Exotic-OMNIBUS) and [here](https://www.rcgroups.com/forums/showthread.php?2711617-Omnibus-F3AIO).

also, feel free to join the communities on [slack](http://185.87.186.121:3000/) or [gitter](gitter.im/betaflight/betaflight)

if this is really a firmware bug, plz reopen :)

closed 2017-04-25 by nathantsoi

avidtech, 2017-04-25:
Thanks,
The Rx is tested working already, and I'm not seeing the resource i2c_scl or i2c_sda declared, so this *might* be firmware specific. I'm going to look around a bit more, already thumbed through quite a bit before posting here.

nathantsoi, 2017-04-25:
ahh, yeah, configurable i2c hasnt been merged yet (
https://github.com/betaflight/betaflight/pull/1867), but it's coming soon
thx to @jflyper's awesome work :)

avidtech, 2017-08-12:
As of RC 3.2, this issue is fixed for the F4 boards.
Issue the following on the command line to enable IIC peripherals on UART3 pins,
`resource i2c_scl 2 b10
resource i2c_sda 2 b11`

I'm putting this information here so anyone who finds this request has an immediate solution.

Thanks everyone!

More information:
#3605