betaflight/betaflight #2944: Revo F4 flashed with wrong FW - now stuck on connect

ilumzar, 2017-04-25:
Hey guys,

made i terrible mistake with my F4 Advanced Flight Controller (MPU6000, STM32F405) and now it´s stuck when i try to Flash it. DFU mode is enabled and immersionRC fixer detects the FC but no matter what FW i try to flash on it, it gets stuck and the "flash" bar is 100% after 1 second..then nothing happens. The usual recovery steps on Betaflight do not work.

AndersHoglund, 2017-04-25:
Turn off "Full Chip Erase".

ilumzar, 2017-04-25:
thx...Ok, did the job. Now, i´m back to my previous issue. BF won´t accept the sbus channel and resets itself to -none-after restart. Tried all UARTS(1-3-6) but they don´t apply as well.

AndersHoglund, 2017-04-25:
What FC ? What target file did you flash ? What version ? Did you select SerialRx on only one UART at a time on the Port tab? Nothing else selected on that UART ?  You did "Save and Reboot" on the Port tab before selecting Sbus on the Config tab?

A lot of questions, but you give us very little to work with. My Revo F4 on BF3.1.7 works just fine, I can select serialRx on any UART. And any serial rx provider, including Sbus.

Then also please remember that a F4 based FC needs special hardware inverter on the UART input for normal sbus to work. This is usually done only on UART1, but you must check your FC how sbus conversion is done. Or use a modified Rx with non-inverted sbus output.

ilumzar, 2017-04-25:
Will give details when back home. Thank you.
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

ilumzar, 2017-04-25:
So, back to the problem

FC: F4 Advanced Flight Controller - (MPU6000, STM32F405)
I flashed the REVO.hex, newest build from this week i guess

> Did you select SerialRx on only one UART at a time on the Port?
  YES and NO, tried all of them
 

> Nothing else selected on that UART ?
 What do you mean?

> You did "Save and Reboot" on the Port tab before selecting Sbus on the Config tab?
 Did that a dozen times withouth success

a friend told me to flash it with raceflight...that an option?

AndersHoglund, 2017-04-26:
What is that "F4 Advanced Flight Controller" ? Link please. Are you sure this is a fully compatible REVO clone?
You can select many functions for a UART. MSP, SerialRx etc etc. You can only have one function selected per UART. And only one UART with SerialRX.

ilumzar, 2017-04-26:
It's the "getfpv" based f4 advanced flight controller. There is no other name for it.
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

AndersHoglund, 2017-04-26:
OK, then there is a thread on RCG. You might get more help there, from other users having this FC.
https://www.rcgroups.com/forums/showthread.php?2779294-Get-FPV-F4-Advanced-Flight-Controller

closed 2017-05-14 by blckmn