betaflight/betaflight #2947: SD Card Shows incorrect freespace size

akira2k1, 2017-04-25:
I have formatted the SD multiple times, but once I plug it into my quadcopter and go to the blackbox tab in Betaflight it says "Free space for logs 4.0 GB" even though its a 32gb SD card with nothing on it.

I'm using
3.1.7 Firmware
Omnibus F4 V3 flight controller
SanDisk Extreme 32gb microSDHC UHS-I UHS Class 3 / Class10

Arakon, 2017-04-26:
FAT32 only supports 4GB files max. As the space is pre-allocated, that's as much as you can get.
That's enough for a few hundred hours of logs, so it should be plenty either way.

jflyper, 2017-04-26:
> *Why only 4GB of my fresh 16GB SDCARD is available for logging?*
>
>You can use every bit (actually less management space) of your 16GB card.
4GB is the limit for a single log session.
>
>As you start using your card, you will continue to see "Free space for logs 4.0GB" message until your card is 3/4 full. After that, the size will decrease as you fill up the rest of your card.
>
>4GB is the maximum size of a single file on a FAT32 file system, and this is how large a single log file can be. The blackbox will try to reserve (pre-allocate) as much space as possible upon sdcard detection, and this space is named "FREESPAC.E". If free space on a sdcard is more than 4GB, then the size of the FREESPACE.E will be 4GB, and configurator misleadingly call the remaining space "unavailable space". Actually, the "unavailable" means "used + free but not used in this log session" in this case. If free space is less than 4GB, then the size of the FREESPACE.E will be the size of the free space, and the rest is "unavailable (used)".
>
>Same applies for 8~32GB cards.

akira2k1, 2017-04-26:
Thanks jflyper! That makes complete sense. I tried doing a search but couldn't turn anything up, thanks.

closed 2017-04-26 by akira2k1