betaflight/betaflight #2952: BF 3.1.7 Tiny Whoop

Jahnkeanater, 2017-04-25:
Pretty sure this is a vibrations issue.

Just updated my Sprecingf3 from cleanflight 1.13 to the latest BF. It flys fine until you pitch forward into fast forward flight. After about 1 second flying forward the quad violently pitches backwards and continues to spiral out of control.

What does cleanflight do differently to handle the vibrations?

Jahnkeanater, 2017-04-25:
flying forward is fine but as soon as you also increase throttle above 50% the quad starts to pitch backwards. I don't think this is just low I term since it only happens in angle mode.

Arakon, 2017-04-26:
This has been reported a few times. 3.1.7 has airmode disabled by default. Enabling it seems to fix this issue.

Jahnkeanater, 2017-04-26:
I also tested cleanflight 2.0.2 and had the exact same issue. I reverted back to CF 1.13 to make sure it is a firmware issue and it flys perfectly again. I have airmode disabled in cleanflight. I'm curious If this is just an issue with vibrations of if it has something to do with the brushed motor code?

MJ666, 2017-04-26:
@Jahnkeanater can you try Betaflight before 3.1.7. we have seen some problems specific to 3.1.7 and small copters. #2852

closed 2018-04-21 by DieHertz