betaflight/betaflight #2954: Possible error in Blackbox numerator/denominator handling

martinbudden, 2017-04-26:
@thenickdude , would you care to comment?

The blackbox logs I-frames and P-frames (for the purpose of this issue report, I'm ignoring other types of frames). I-frames contain absolute values and P-frames contain delta values. By default every 32nd frame is an I-frame, so we get an I-frame, 31 P-frames and then another I-frame. Eg
```
0         1         2         3         4         5         6         7
01234567890123456789012345678901234567890123456789012345678901234567890
IpppppppppppppppppppppppppppppppIpppppppppppppppppppppppppppppppIppp

```
[P-frames in lower case to make the stream easier to read]

However, if the flight controller is running at a high rate (eg 8kHz) then we can get dropouts due to the recording medium not being able to handle the data rate. To solve this problem, the blackbox can be set to run at a slower rate. So if `blackbox_rate_num` is `1` and `blackbox_rate_denom` is `4` then every fourth frame will be logged.

The possible error is in what is meant by "every fourth" frame. What actually happens is that every 32nd frame is an I-frame and 3 in 4  P-frames are skipped, ie
```
0         1         2         3         4         5         6         7
012345678901234567890123456789012345678901234567890
I...p...p...p...p...p...p...p...I...p...p...p...p...p...p...p...I...p...p...p...p...p...p...p...
```
That is every 32nd frame is an I-frame and there are 15 P-frames between each I-frame.

What I think is a better behaviour is to retain the I-frame to P-frame ratio, so instead every 64th frame is an I-frame and there remain 31 P-frames between each I-frame, eg
```
0         1         2         3         4         5         6         7
012345678901234567890123456789012345678901234567890
I...p...p...p...p...p...p...p...P...p...p...p...p...p...p...p...I...p...p...p...p...p...p...p...
```
Note the difference, the frame at position 32 (capitalised for clarity) is now a P-frame instead of an I-frame.

The reason for this include:
1. I-frames are larger than P-frames, so retaining means size savings in the data stream are not maximised
2. The frame rates are higher, so the we should not be concerned with the time delta between I-frames
3. Having an I-frame fixed at every 32nd frame makes logging FCs running at high rates difficult. So if the FC is running at 32kHz, then with `blackbox_rate_denom` at 32 we get an I-frame every 1ms and no P-frames. Whereas with an FC running at 1kHz and `blackbox_rate_denom` at 1, we get an I-frame every 32ms with a P-frame on each 1ms interval.

ledvinap, 2017-04-26:
@martinbudden: IMO the goal is to have I frames with roughly constant intervals and reduce bandwidth by skipping P frames. So when running at 2Khz, I frames should be 63 p frames apart. blackbox_rate_num / blackbox_rate_denom will then work as expected.

martinbudden, 2017-04-26:
@ledvinap , from my reading of the code, I-frames are produced every 32nd call to `handleBlackbox` regardless of PID loop rate.

ledvinap, 2017-04-26:
@martinbudden: Yes. But it was written when 1kHz was nearly only update frequency used and bandwidth (openlog) was quite small.
Now faster update rates are possible and common, so blackbox should take that into account.

martinbudden, 2017-04-26:
Yes, that was the point I was trying to make.

martinbudden, 2017-04-27:
@ledvinap , how about this for proposed defaults:

PID rate | I-frame interval | I-frame ratio | P-frame interval | P-frame ratio | S-frame interval | S-frame ratio
-- | -- | --  | --  | -- | -- | -- 
500Hz | 32ms | 16 | 2ms | 16 | 4096ms | 64
1kHz | 32ms | 32 | 1ms | 32 | 4096ms | 128
2kHz | 32ms | 64 | 1ms | 32 | 4096ms | 128
4kHz | 32ms | 128 | 1ms | 32 | 4096ms | 128
8kHz | 32ms | 256  | 1ms | 32 | 4096ms | 128

The I-frame ratio is the number of PID loops per I-frame. The P-frame ratio is the number of P-frames per I-frame. The S-frame ratio is the number of I-frames per S-frame (inverted compared to other two ratios). By default the GPS frame would be sent at the same frequency as the S-frame.

That is the defaults are set so that an I-frame is sent every 32ms and a P-frame is sent every 1m, regardless of PID rate.

Additionally there would a navigation setting where I-frames, G-frames and S-frames were all sent once per second, with no P-frames.

ledvinap, 2017-04-27:
@martinbudden : I like this proposal, it should cover most use cases cleanly and with enough resolution.

- Number of P-frames per I-frame should be 31 (I-frame has all data from P-frame, you want 32 frames per interval)

- Same math will be necessary to handle cases with non-integer `I-frame ratio/P-frame ratio`. Some rounding error will result when I-frame is inserted ('fractional part' of P-Frame counter is discarded), in real use cases user should avoid it anyway.

- Maximum usable value of I-frame ratio is larger than 256. `uint16_t` is easiest, 8 bits will be enough with some floating-point like magic.

martinbudden, 2017-04-28:
@ledvinap , thanks for comments.

1. Yes, 31 vs 32 was just me in the habit of typing 32. Clearly 31 P-frames per I-frame.
2. I was thinking we could only allow integer I-frame P-frame ratio. I know it handles the fractional case now, but I don't think it is really necessary.
3. I was going to use a `uint16_t`

martinbudden, 2017-04-28:
@ledvinap , @thenickdude,

I've also been thinking about what is included in the P-frame. Currently the P-frame includes slow-changing data, such as battery voltage and current, RSSI, baro etc. I propose these are only included in the I-frame.

So the P-frame only includes the following:

time, PIDs, rcCommands, gyro, acc(if enabled), debug(if enabled), motors

thenickdude, 2017-04-28:
Including the precise moment that variables change was intended to allow
the debugging of flight behaviour (which input is the flight controller
reacting to? Is the reaction correct? Particularly useful for threshold
behaviour like battery voltage). And with the differential encoding it's
very cheap to log unchanged fields...

martinbudden, 2017-04-28:
@thenickdude , that makes sense. I'll leave those fields in then.

closed 2018-04-21 by DieHertz