betaflight/betaflight #2959: Feature Request: GPS Dynamic platform model CLI command

Phanny-P, 2017-04-26:
When using GPS auto config, I see that the default dynamic platform model (ublox) is set to the Pedestrian mode: "Applications with low acceleration and speed, e.g. how a pedestrian would move. Low acceleration assumed." I was able to compile a hex to have the auto config enable the "airborne <2g" dynamic model with great results. Maybe there was an easier way to enable it(?), but I am new to this.

Could the "airborne <2g" be set as the default with auto config? Or better yet, have a CLI command to switch dynamic model modes? Thanks.

McGiverGim, 2017-04-26:
Do you have seen some improvement? According to ublox documentation:
```
Pedestrian: Applications with low acceleration and speed, e.g. how a pedestrian would move. Low
acceleration assumed. MAX Altitude [m]: 9000, MAX Velocity [m/s]: 30, MAX Vertical
Velocity [m/s]: 20, Sanity check type: Altitude and Velocity, Max Position Deviation: Small```
The airbone mode can have more m/s, but the position deviation is large. 
20 m/s is about 72km/h, and 30 m/s is 108km/h, so maybe it's better let configure it because some quads are really faster than that.

Phanny-P, 2017-04-26:
Yes, I have had **significant** improvement. I have read the ublox M8 documentation and have seen the note on the position deviation. However, GPS velocity is based off of Doppler calculations - not position. This is why the positional accuracy for the ublox M8N is 2.5m, but the velocity accuracy is 0.05m/s. Not many people realize this and it is the reason many think radar guns are more accurate. I am just concerned with accuracy of the velocity, not position.

You also brought up another great point - the max velocity is lacking for the Pedestrian mode. The ublox description: "e.g. how a pedestrian would move." doesn't correlate very well to quadcopters :)

Phanny-P, 2017-04-26:
Not sure if I can post a link, but here is a video of the osd speed (in ft/s). I also drove my quad around in my car - velocity is very responsive and accurate.
https://youtu.be/CLArfxTj15M

mikeller, 2017-04-26:
@Phanny-P: That sounds like a great improvement. Can you please submit a pull request for this change?

Phanny-P, 2017-04-27:
@mikeller: Yes, I will read up on how to do it - it may take me a few days due to time constraints. My background regarding GitHub and how it all works is completely new for me (my background is mostly mechanical). This hobby is extremely inviting to all engineering disciplines and I am loving it :)

mikeller, 2017-04-27:
@Phanny-P: No problem, take your time, any contribution is appreciated. We're also always happy to help, if you have questions or want to bring up something for discussion, feel free to join the Betaflight slack by going to http://www.betaflight.ch/ .

Phanny-P, 2017-05-18:
@mikeller : I finally got around to doing this - this is very foreign to me and I hope it was done correctly...

dk7xe, 2017-09-18:
is the GPS airborne <2g model already implemented and if in which version? 
My OSD is showing <90km/h while i'm flying >130km/h . GPS is a M8N in my case. BF 2.1.7
See https://youtu.be/b3z7slZ9GjE

mikeller, 2017-09-18:
@dk7xe: Yes, #3110 was merged a while ago. But not in time for 3.1.7, you'll have to update to 3.2.0 (currently in RC).

dk7xe, 2017-09-18:
thanx @mikeller. i will update to 3.2.0RC the coming days.

dk7xe, 2017-10-11:
I can confirm that GPS is working correctly with BF3.2.0 RC5 now.
See https://youtu.be/L8aQNGfyqNk