betaflight/betaflight #2960: Beecore frsky 3d?

Dracen-Knights, 2017-04-26:
ok so I picked up a tinywhoops style quad to get more stick time while I'm down for my next surgery recovery... well long story short I have 5 other quadcopters all setup for 3d flight ad I want to set up this one to do the same. The main reason is that I have my taranis xd9 plus set up fo both sticks selfcenter, and I din't want to change that for this one quad. I noticed that BF 3.1.2 (which came on it) has the 3d mode slider but it doesn't save that when turned on and click save and reboot. so is it possible for the beecore frsly to be able to use 3d mode? I know the motors are pwm and the motors can spin either directions just not sure the FC can do it or how I would go about setting it up. Any way if this FC can't do it whats my other options for getting a 3d tinywhoops?

brucesdad13, 2017-04-26:
These little brushed motors don't like to be run in reverse.

https://community.micro-motor-warehouse.com/t/reversing-motor-rotation-for-better-flight-characteristics/3158/3

Dracen-Knights, 2017-04-26:
I'm not worried about them liking it or shortening the life span I just want to be able to by simply pushing the stick down instead of having to change my controller. I'll look at the link and see what I can find.

brucesdad13, 2017-04-26:
I'm curious to know what you find out :)

MJ666, 2017-04-26:
Brushed motors even can not run in reverse with the usual brushed ESC's.

AndersHoglund, 2017-04-26:
A brushed DC motor direction is controlled by voltage polarity. The usual brushed FC/ESC has a fixed polarity with only one FET switch controlling power. No way to reverse other than by swapping wires.
You would need four FET switches in a bridge configuration to have a reversible brushed DC motor, I think some car ESCs have this.

Dracen-Knights, 2017-04-26:
Ok the fact they have one FET is what I was wondering and looking for so time to either figure out a way to add more or time to design and build a micro brushed quad. ... I was really hoping that it could do it but guess it's ok I'll just have to learn to fly this one holding the throttle stick down as I'm not going to mess up the way I am used to with my main quads just so I can fly one indoors...

reedbn, 2017-05-02:
@Dracen-Knights you can't do negative RPM/collective, but you could potentially set your transmitter to map the centered stick to 1000 and full throttle to 2000. That'd cut your total throw in half, but might mimic the 3D control feel better?

fftunes, 2017-05-03:
Yes, adjust transmitter throttle curve so center stick is 1000.

closed 2018-04-21 by DieHertz