betaflight/betaflight #2964: SP Racing F4 Evo: Alt Hold jumping up and down

netzwergracer, 2017-04-27:
SP Racing F4 Evo Flight Controller's Altitude Hold mode is unusable, due to the fact that the quad jumps up and down, oscillations getting stronger exponentially resulting in crash.
Altitude PID Settings don't seem to stop it, tried all kinds of combinations.
PS: What does VEL below ALT PIDs do?

Arakon, 2017-04-27:
Did you cover the baro with foam?

unitware, 2017-04-28:
I have this probelm as well, but it works rather ok when moving. (I have not covered the baro in foam, it came loose.. will do that again)

netzwergracer, 2017-04-29:
Yes, baro is covered by big piece of foam.
Baro logs in blackbox look good too, very even graph.

mikeller, 2017-04-29:
Is ACC enabled? Data from ACC is used to improve the accuracy of the baro data.

netzwergracer, 2017-04-30:
Yes, accelerometer is enabled too.

closed 2018-04-21 by DieHertz