betaflight/betaflight #2965: LED strip with Lumenier MICRO LUX F4 Flight Controller

starz91, 2017-04-27:
Hi,
I can not run the led strip with this FC (micro lux lumenier). The leds remain white.

With a piko blx no problem.

**Here is the reference of led :**
DIATONE SW316/AN Full Color Flash Bang 2812 LED Board 5V Input Direction Night Light Plate For Racer

--> BANGGOOD

Can you help me ?

AndersHoglund, 2017-04-27:
Have you done the "series diode" mod to lower supply voltage? Read all about it: 
https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md

jr3310, 2017-05-06:
Hi,
I too am having issues with the 4 LED strip not working on my Lumenier Micro LUX flight controller. The LED was working initially until I flashed to the latest firmware, Bluejay 3.1.7. I have 5 volts showing at the LED connection and have reprogrammed the LED configuration several times. I have even reinstalled the original firmware version and replaced the 4 LED strip but that didn't fix it either. The LED strips work fine on my old flight controller. Also, I can't seem to find any documentation on this flight controller anywhere. 
Anyone have first hand experience with Lumenier LED issue?

mikeller, 2017-05-07:
I have had a quick look at this board, but I was not able to generate any DShot / LED_STRIP output for the pin labelled 'LED_STRIP', so it stands to reason that this pin potentially isn't properly hooked up to an MCU pin on the Lumenier micro LUX board. Probably open a support request with your supplier?

jr3310, 2017-05-07:
Hi,
The LED was working initially until I flashed to the latest firmware, Bluejay 3.1.7. and configured for Dshot 600.

mikeller, 2017-05-08:
@jr3310: What firmware version were you using before you updated to 3.1.7?

jr3310, 2017-05-08:
I'm not sure what version I had before I updated. I do know that Dshot was not an option for the ESC on the factory firmware. I tried to go back to BLUEJAY ver 3.0.1 but the LED still wouldn't work.I'm back to BLUEJAY 3.1.7 firmware now.

mikeller, 2017-05-10:
Hmm, that's odd, I can't find a version of the BLUEJAYF4 target without DShot support. Looks like their board comes with a custom built target. This means that the LED pad is potentially connected to a pin that isn't even used in the original BLUEJAYF4 target.

Hang in there, I will see if I can get information from Lumenier on how the LED pin on this board is connected.

Biofreezer, 2017-08-23:
I had also an LED_STRIP issue with the Lumeniere Micro Lux F4. As soon I flash to an new version 3.1.7 BLUEJAYF4 the LED stopped working. I flashed the 3.0.1 BLUEJAYF4 (with no DSHOT) and added the dump file to CLI:[Dump-Original.txt](https://github.com/betaflight/betaflight/files/1244227/Dump-Original.txt) and the LED started to work again.

jflyper, 2017-08-23:
@Biofreezer You have to find out which MCU pin is connected to the LED_STRIP pad/TH.

Biofreezer, 2017-08-23:
Its PIN43 on the ARM2 STM32F405RTG6

mikeller, 2017-08-24:
@Biofreezer: That's the UART1_RX pin on the regular BLUEJAYF4 board. This will probably require a custom build to work with Betaflight (just like the custom 3.0.1 build that Lumenier supplied). Can you try if the attached firmware works please?

[betaflight_3.2.0_BLUEJAYF4.zip](https://github.com/betaflight/betaflight/files/1247585/betaflight_3.2.0_BLUEJAYF4.zip)

jflyper, 2017-08-24:
@Biofreezer 
> Its PIN43 on the ARM2 STM32F405RTG6

This can't be true. This pin (PA10) is, as @mikeller pointed out, is UART1_RX, which is the fundamental part of the board. If this pin is connected to LED strip, then the board is not compatible with Bluejay F4 at all.

You probably have identified the pin with incorrect orientation of the chip. The chip orientation is determined by "Pin 1 marker", which is a dimple on the corner, NOT how marking is printed.

[image]

mikeller, 2017-08-24:
Not sure of that. Lumenier supplies them with a custom firmware, which should have could have UART1 disabled, since there are no pins for it: http://d1b5yzykmxw6e8.cloudfront.net/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/m/i/micro-lux-drawing_1.jpg

jflyper, 2017-08-24:
Oh, okay ...

Here's are the possibilities.

[image]

I suspect it's on PB1 (or may even be on PB0).

mikeller, 2017-08-24:
Aaand the winner is:

PB1!

`resource led_strip b1` will make your LEDs all shiny.

jflyper, 2017-08-24:
Oh...

BTW, is that with 3.1.7? Or 3.2?

Biofreezer, 2017-08-24:
[image]

Pin43 = PB1

jflyper, 2017-08-24:
@Biofreezer 
We are done with this issue. Like @mikeller noted,
> `resource led_strip b1` will make your LEDs all shiny.

-------
FYI, I think you are really confused.

1. First, the pin under question is not 43, but 27.
You are probably looking at the chip with surface marking right side up.

[image]

2. Second, pin 43 is not PB1, it is PA0 (this is a minor problem).
[image]

Biofreezer, 2017-08-24:
Yes, you are right, sorry my bad. Thanks for the explanation.

TheMightyThor1212, 2017-09-13:
Hello

I realize this thread might be a little dated. from reading it, it appears to be a resource mapping issue? is that correct? just remap the LED to B1 (resource led_strip b1) and it works by connecting to the LED Connection?
thanks

mikeller, 2017-09-13:
@TheMightyThor1212: Read https://github.com/betaflight/betaflight/issues/2965#issuecomment-324549233: Yes.

TheMightyThor1212, 2017-09-14:
@mikeller thank you so much!!!!

TheMightyThor1212, 2017-09-14:
@mikeller woukld anything else have to get remapped, say maybe the DMA Channel? i can get the LEDs to show Color but it doesnt respond to larsen scanner, only the first light comes on, the other 3 blink rapidly. didnt know if there was more to remap. i am using the latest firmware for the MICRO LUX F4, bluejay target. thanks in advance!!

mikeller, 2017-09-15:
Hmm, weird, it worked fine for me. Can you please make sure that nothing is configured for UART1 on the ports tab?

TheMightyThor1212, 2017-09-15:
@mikeller Ok i will look again, off the top of my head i can remember, but thank  a lot for responding its greatly appreciated!!

TheMightyThor1212, 2017-09-15:
@mikeller nothing configured on UART1, it is off.

TheMightyThor1212, 2017-09-15:
@mikeller maybe its something to do with the firmware, if i make the lights white, they all turn on. if i change Light 0 to a color or Larsen scanner, only light 0 blinks. probably a mapping issue or something is my best guess. nothing else works, no color changes, no indicators, etc. just white lights. i tried a second brand new strip, same results. i ordered more LED Strips, maybe those will work. not important but it wold be nice to have the LEDs. thanks for all your help.

jflyper, 2017-09-15:
@TheMightyThor1212 
I suspect it is the voltage issue. How are you powering your LEDs?

TheMightyThor1212, 2017-09-15:
@jflyper 5V power from the LUX. i measured, it is 5 v

jflyper, 2017-09-15:
@TheMightyThor1212 
Hmm... why don't you make it a bit lower, just in case, by inserting a diode.

If this doesn't work, why not a diff output so @mikeller can load it onto his board.

TheMightyThor1212, 2017-09-15:
@jflyper I'll try a few things, these are all 5V strips, they all work on other aircraft, all lights come on when they are programmed to be white. but i am game for anything. not sure what you mean by a DIFF output? are you saying use a different connection on a different port? or are you asking me to do a dump?

AndersHoglund, 2017-09-15:
Problem with LED-strips are not the +5V supply as such. The problem is the difference between input signal high level and power supply voltage. Too big difference and it will not work, easiest way to make this difference a bit smaller is to lower the power supply voltage with a diode in series. Amplifying the input signal would be an alternative, a bit more circuitry needed though.
And jflyper is not asking you for a full `dump`, only to do a `diff `command.

TheMightyThor1212, 2017-09-16:
Thanks @AndersHoglund never did a diff command, so i didnt understand it. i did a resource command @jflyper  @mikeller it seems that B01 is assigned to the LED_STRIP, MOTOR 6 and a SERIAL_TX 11. if i do a resource list, i do not see MOTOR 6, but i DO see Serial_TX 11, which i assume is the LED channel? I have always run these LEDS, which i think are 2811s? on a 5v connection, though they can run on a 3.3V. i do not think Micro LUX F4 has a 3.3 volt connection? i thought that was a problem i read about a while back that spectrum users couldnt use the Micro Lux F4, maybe i am confusing it with another Lumenier Board, like the AIO. so thats where i am at.

TheMightyThor1212, 2017-09-16:
HERE is my DIFF command results
# diff

# version
# Betaflight / BLUEJAYF4 3.1.7 Apr  3 2017 / 21:54:04 (e1c4b5c)

# name
name Frik

# resources
resource MOTOR 1 A01
resource MOTOR 2 A00
resource MOTOR 3 A03
resource MOTOR 4 A02
resource LED_STRIP 1 B01

# mixer

# servo

# servo mix

# feature
feature -VBAT
feature -BLACKBOX
feature -SDCARD
feature LED_STRIP

# beeper

# map

# serial

# led
led 0 3,0::CI:1
led 1 2,0::CTO:6
led 2 1,0::CTO:6
led 3 0,0::CI:4

# color

# mode_color

# aux
aux 0 0 1 1700 2100
aux 1 28 0 900 1300
aux 2 1 0 1675 2075
aux 3 2 0 1250 1700
aux 4 13 2 1375 2075

# adjrange

# rxrange

# rxfail

# master
set motor_pwm_protocol = MULTISHOT
set small_angle = 180
set gyro_notch1_hz = 0
set yaw_motor_direction = -1
set pid_process_denom = 1

# profile
profile 0

set d_lowpass_type = PT1
set setpoint_relax_ratio = 25
set d_setpoint_weight = 190
set p_pitch = 45
set i_pitch = 40
set d_pitch = 44
set p_roll = 35
set i_roll = 30
set d_roll = 42
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 130
set rc_rate_yaw = 115
set rc_expo = 30
set rc_yaw_expo = 30

#  @AndersHoglund  @jflyper @mikeller

mikeller, 2017-09-16:
@TheMightyThor1212: @jflyper and @AndersHoglund are right, the symptoms you describe sound totally like a voltage issue. Add the diode to the Vin of the LED strip, and you'll be fine.

TheMightyThor1212, 2017-09-16:
Thank you all for your help, before i do that i will wait on the new LEDs to come in. these arent the greatest LEDs known to man, but they have worked on other FCs, like the PIKO BLX. probably time for a new FC anyway. @AndersHoglund  @jflyper @mikeller

mikeller, 2017-09-16:
As @AndersHoglund tried to explain. It is not that your LEDs are not good. It is that driving them at 5V, with a signal level of 3.3V is touch and go. It will work if the flight controller's 5V rail is slightly low, or the 3.3V rail slightly high, but not if it is the opposite way. From your symptom description it looks like the flight controller you are using does not meet these criteria, so using a different set of LEDs without a diode will not fix this.

TheMightyThor1212, 2017-09-22:
Ok I figured it out. I had inadvertently connected the LEDs to the 3.3V when it should have been connected to the 5V. Once i connected to 5V it worked perfectly. thats what happens when you flip the board upside down :P again, thanks for all your help, i just wanted to leave this comment here incase some one else comes looking for the same reason!! @mikeller @AndersHoglund @jflyper

closed 2018-04-21 by DieHertz