betaflight/betaflight #2967: Strange difference between debug and non debug build.

MJ666, 2017-04-27:
We see some strange issue during testing the AlienFlight F7. If I do a normal build the resulting firmware has inverted signal output for the motor 2 port. If I do an debug build everything is working normally for the motor 2 signal. Have verified this with a scope. Does someone has a idea what's going on? This is really strange. For now we continue testing with the debug build.

brucesdad13, 2017-04-28:
Which compiler/toolchain are you using MJ?

MJ666, 2017-04-28:
Using the actual GNU ARM 'tool chain with eclipse.

MJ666, 2017-04-29:
The debug build is much slower and is not running well with 8K/8K anymore. I now only build pwm_output.c in debug mode and this makes the motor 2 port also working normally. I'm just wondering if the optimization used for the non debug build is kicking some important code? I'm now a bit scared this also could happen at other places. I tried to compare the generated assembler code for pwm_output.c but in non debug build I only get an array of binary data? Does someone have an idea how to go forward with this and solve the issue?

ledvinap, 2017-04-29:
@MJ666 : debug information is available even with higher optimization level, -O0 only makes debugging easier. -O1 is optimization level intended for debugging - optimizations that interfere with debugging are disabled.
Binary data you see are probably caused by -flto, only intermediate code representation is emitted in 'compiler' phase, 'linker' pass is responsible for actual compilation (all source files together)

gcc is very mature compiler. If there is difference between optimized and unoptimized code, it is almost certainly caused by bug in code (incorrectly using memory that is modified in background / timing constraints / using construction with undefined behavior)

MJ666, 2017-04-29:
Motor 1 and 2 are really configured in the same way and coming from the same timer. I now have checked all motor outputs for the signal in more detail. Motor 7 output is also inverted. I use all 4 channels of timer 8 for motor outputs. Channel 2 and 3 are inverted in the non debug build. Channel 1 and 4 behave normally.  Not sure why they behave differently in debug and non debug build. I also noticed there is some minor delay in the motor output between different timers even in synchronized mode.

MJ666, 2017-04-29:
@ledvinap will see what I can find during debugging of the non debug build.

mikeller, 2017-04-29:
@MJ666: I am having the same problem with an F7 test board that I am working on. Motor outputs on TIM8 and TIM1 are inverted. If I change the pins to use different timers, they are not inverted. (Haven't tried debug build yet.)

sambas, 2017-04-29:
TIM1/8 are advanced mode timers, code might be missing something important for them

MJ666, 2017-04-30:
I tried to debug the default build version but for my taste there is quite a lot optimized out wich should be still available. Looks really strange. The firmware version with the debug build of pwm_output.c was flight tested last days and is behaving quite well at a brushless copter and Multishot.

We also found out we have some issue with DSHOT support. Even if we have an unique DMA cannel assigned for each motor port in resource list not all motors work with DSHOT.  When SDCard DMA and VBAT/CURRENT_METER is disabled we have an DMA channel available for each of the 8 motor ports. For the first 4 motors the actual problem is at motor 3 but we have not really checked all of them yet.

mikeller, 2017-04-30:
@MJ666: Could it be a case of the DMA channel that is not working being used for SPI or something else? Betaflight does not take this into account or display it.

ledvinap, 2017-04-30:
@MJ666 : It takes some time to get used to debugging highly optimized code ... 
Try reducing optimization level to -O1 if problem persists.

Or you may try adding busy loops into initialization, maybe you will discover path with critical timing

ledvinap, 2017-04-30:
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_output_stm32f7xx.c#L207 TIM_OCInitStructure is uninitialized and only part of the fields is set.

MJ666, 2017-04-30:
@mikeller In the meantime I have checked with the scope and dshot signal is there but also inverted. be havior of all 8 port is different in debug and non debug mode.
@ledvinap I will check the programming guide and see if I can find what is missing. Thanks.

mikeller, 2017-04-30:
@MJ666: Can you test #2992 please? I will test once I'm home tonight.

MJ666, 2017-04-30:
More detail at #2992 I have tested the following fix successful. it may be still not 100% correct. https://github.com/AlienWiiBF/betaflight/commit/5bfe52e997f17f7b9bfe6ebd973d4e251c288a1a

@ledvinap have not found code to initialize the TIM_OCInitStructure. In various examples I found there is no initialization between defining the structure and assigning values to elements.

ledvinap, 2017-05-01:
@MJ666 : `memset` to zero the structure may be enough. It is allocated on stack, so it has undefined content. 

ST code assigns all fields when TIM_OC_InitTypeDef  is used, but that sounds quite fragile to me.

mikeller, 2017-05-02:
@ledvinap: That sounds reasonable. @MJ666: Do you want to try this and submit a PR if it works?

MJ666, 2017-05-02:
Even if the problem itself is solved with the actual master I created #2997 to get the TIM_OCInitStructure structure completely initialized when used.

closed 2017-05-02 by MJ666