betaflight/betaflight #2969: Fc keep forcing to DFU mode

dbrokenbone, 2017-04-28:
couple days ago i land on grass with alot of dew. after that, every time i change battery it force me to DFU mode. i hv to pulg unplug many times to reboot properly, but i check the boot button isn't short. any idea where i should searching for?or maybe turn off DFU mode...

mikeller, 2017-04-28:
DFU is hard coded into the MCU, so it's not possible to turn it off. If it's not the button, it could still be a short between boot0 and Vcc, caused by a bit of grass stuck to the board. I recommend the usual treatment for electronics after contact with water: Remove battery, wash the board down with isopropyl alcohol (or any other kind of alcohol), make sure you remove any foreign objects stuck to it, thoroughly dry in a warm spot.

dbrokenbone, 2017-04-28:
thanks a lot...i'll try it with wd40

Arakon, 2017-04-29:
WD40 is NOT suitable for this purpose.

dbrokenbone, 2017-04-29:
any reason?

Arakon, 2017-04-29:
It leaves an oily residue that attracks tons of dirt and can make plastic parts (i.e. IC cases) brittle. Isopropyl is the proper way, since it evaporates almost without any trace.

dbrokenbone, 2017-04-29:
thanks alot...its very helpful..

closed 2017-04-29 by dbrokenbone

mikeller, 2017-05-01:
@Arakon: Totally agreed. I did not want to be as specific, as isopropyl may be difficult to obtain in certain places. Any kind of alcohol, like methylated spirits, should do an ok job.