betaflight/betaflight #2970: ESP32 + (GY91) MPU9250 BMP280

triblax, 2017-04-28:
Hello, you're thinking about porting to use ESP32 + (GY91) MPU9250 BMP280?

jflyper, 2017-05-02:
Not likely.

It's going to be a substantial work to port BF to non-STM32 architecture.

cs8425, 2017-05-02:
I think it is possible, but need lots of work to port low level stuff.
And maybe need to consider timing/jitter issue.

nathantsoi, 2017-05-02:
I agree, tons of work @triblax, but it would be awesome to abstract the core PID loop and drivers out so we could use this on any MCU. Also the Cypress psoc has some potential w/ the integrated cpld/analog circuitry and fully reassignable pinout

stealthhack, 2017-05-03:
I am willing to donate if you want to work on porting for ESP32, afterall the board is about 5$ or less, dual core, and even CAN bus for future work with communincation!

cs8425, 2017-05-03:
I had taken a look about coding ESP32.
It's base on FreeRTOS (only have the FreeRTOS SDK currently),
so needs wrapper to wrap BF code.
And scheduler parts (almost) needed to be rewrite.
That's a hard work.

triblax, 2017-05-03:
I found, that esp32 can be programmed by modified Arduino IDE/SDK. But I don't know how deeply low level...

stealthhack, 2017-05-03:
This guy made a pretty good work on Arduino with the old esp8266:
http://www.brokking.net/ymfc-al_main.html
The arduino code is on the site, it can be downloaded and matched for esp32 accordingly.

bjoern79de, 2017-08-06:
I'll try it. I've already bound the SPI bus and got the MPU9250 running and I'm able to connect the configurator.

jflyper, 2017-08-10:
Points I'm scared of beside rewriting entire driver:

1. There are arguments on extremely low floating point performance of the ESP32 core on the net.
https://blog.classycode.com/esp32-floating-point-performance-6e9f6f567a69
https://www.esp32.com/viewtopic.php?t=800

2. Lack of DSP instructions.
Dynamic notch is really handy. Are we going to implement the number cruncher with the second core?

closed 2018-04-21 by DieHertz