betaflight/betaflight #2973: BF 3.1.7 / SPRACINF4EVO -  KISS ESC pass-through feature not available

dk7xe, 2017-04-28:
FC - SPRACINGF4EVO
BF version - 3.1.7
I tried to flash my KISS24RE on the "[Monster](https://goo.gl/photos/N8D8LjbBLehtDhnA8)" according [Betaflight ESC pass through](https://github.com/betaflight/betaflight/wiki/Betaflight-ESC-pass-through) instruction.
When entering 
`escprog ki 255` 
via CLI i get a error message that this command is not supported.

Please add  KISS ESC pass-through functionality for SPRACINGF4EVO.

dk7xe, 2017-05-31:
one month passed now with no reaction so far ;(

mikeller, 2017-05-31:
@dk7xe: Yeah, we could really do with some additional developers! ;-)

mikeller, 2017-06-02:
Also, having a quick look at the target definition, it seems to be that `ESCSERIAL` is commented out, probably because it had issues when 3.1.7 was released. (@hydra would know)

Attached a build of 3.2.0 with `ESCSERIAL` enabled, if somebody wants to test if it is working.
[betaflight_3.2.0_SPRACINGF4EVO.hex.zip](https://github.com/betaflight/betaflight/files/1046617/betaflight_3.2.0_SPRACINGF4EVO.hex.zip)

dk7xe, 2017-06-02:
Hi Michael, thank you! I will give it a try the coming days.

dk7xe, 2017-06-28:
Did the test this morning with the FW @mikeller provided. In principle it is working BUT only ESC of motor 1,8,4 and 3 of the Octo Flat X were flashed.
Is there anything i need to do or is that a general issue?

mikeller, 2017-06-28:
Not sure. I suspect this has never really been tested for more than a quad. Did you do the flashing ESC by ESC, or did you use `255` for 'all ESCs'?

dk7xe, 2017-06-28:
Damned.. i guess the problem is in front of the computer. I did use 255 but since i was in a hurry stupid me has forgotten to change the configuration to "Octo Flat X" before flashing.
Will give it another try tomorrow morning.

mikeller, 2017-06-28:
:-)

dk7xe, 2017-06-29:
The problem was indeed in front of the computer.. as usual. After changing the configuration to `Octo Flat X` before flashing the `escprog ki 255` worked flawless on all of the 8 ESC's.
Thank you @mikeller !

closed 2017-06-29 by mikeller

dk7xe, 2017-06-29:
And finally the video https://youtu.be/b8OhpH2uN1M
;)