betaflight/betaflight #2975: Cannot Flash F3 FC

resqmedic, 2017-04-28:
Unable to flash the F3 board. When started in bootloader port switches to DFU and haven't really found anything you can do when using Mac.

jflyper, 2017-04-29:
I've NEVER had ANY sort of flashing problem with my Mac.

Let us see a screen shot of your flasher.

resqmedic, 2017-04-29:
I will when I get home much later today. Can you give me the exact process to make sure I'm doing it right?

HackIT, 2017-04-29:
Hey, Put the board in DFU(3 type of board... some uses silab's USB to UART some use stm32's DFU and some have an SPI debug port, without the skill you can't understand what I'm speaking about)... plug it then look at the select menu top right from betafligh... you shud see your new device, at this moment you know it is recognized... so you can flash it... if it doesn't recognize your board! you have to find how to add your new device to your system so betaflight can see it...

resqmedic, 2017-04-30:
On a mac. When I connect the board without holding the bootloader button, it shows up as /Dev/cu.usbmodemfFA131   When I hold the bootloader button and connect, it changes to DFU and the previous port isn't in the drop down list

jflyper, 2017-04-30:
Yes, that is the way it should be.

Select the firmware flasher icon and proceed.

resqmedic, 2017-04-30:
I load the firmware, click flash, and nothing happens

resqmedic, 2017-04-30:
[image]

resqmedic, 2017-04-30:
[image]

jflyper, 2017-04-30:
@resqmedic Can you expand the log ("Show log")?

resqmedic, 2017-04-30:
[image]

jflyper, 2017-04-30:
Any additional log when you try to "Flash Firmware"?

resqmedic, 2017-04-30:
Nope, that's after I click flash firmware

resqmedic, 2017-04-30:
When i try to connect,
[image]

resqmedic, 2017-04-30:
Also same thing with Cleanflight

Wtcheesecakes, 2017-04-30:
can you take a picture of your flight controller // is it definetley a betaflight f3 board (from boris)

resqmedic, 2017-04-30:
Its in the quad all soldered up at the moment. Ordered from Racedayquads.com. They are also trying to help me via email. I noticed something as well, when i try to connect to the board, it stays in "waiting for data." Took out two old naze boards, were able to flash them both, but when i connect to them "waiting for data."

sambas, 2017-04-30:
have you tried flash with full chip erase?

jflyper, 2017-05-01:
At this point, I also suspect that your board is not BetaflightF3.

Do you have anything that came with the board? A box, a pin map sheet, a document, anything that can be used to identify the board.

HackIT, 2017-05-01:
hmmm, the last screenshot is done when the board is DFU or without the boot pin/button pressed?
If it's not DFU then the board doesn't work... if it's DFU it's normal that you can't connect to the application running into the flight controller...
Maybe try another usb wire...

resqmedic, 2017-05-01:
@jflyper 
[image]

resqmedic, 2017-05-01:
Went out and bought another usb cable, still nothing. The bottom bar will say "loaded online (also tried local) firmware. The port shows DFU, when I click flash, nothing happens other than the log saying "USB Device Successfully opened with ID:1.

resqmedic, 2017-05-01:
[image]

mikeller, 2017-05-01:
Now, from this screen, check 'No reboot sequence', and then click 'Flash'.

resqmedic, 2017-05-01:
[image]

mikeller, 2017-05-02:
The 'Loaded Local Firmware...' in the status bar at the bottom makes it look like you never clicked 'Flash Firmware'.

resqmedic, 2017-05-02:
It stays that way when I click flash firmware

mikeller, 2017-05-02:
Ok, that's odd. Try removing the configurator from Chrome, and reinstalling it.

resqmedic, 2017-05-03:
Did, I ordered a new board outta frustration, if it works I'll let everyone know, it's gotta be the board.

mikeller, 2017-05-03:
Pretty sure it's not the board, if 'DFU' is displayed at the top right, the board goes into DFU mode, and flashing should work.

jflyper, 2017-05-03:
@resqmedic Did you try different USB port on your Mac?

HackIT, 2017-05-03:
Give it a try on another computer... if you can. Mac can be problematic as it seems...

resqmedic, 2017-05-03:
@HackIT bingo we gotta winner.
Turns out my MacBook just doesn't like betaflight. Used a windows laptop with impulse rc driver and did it in about 5 minutes. Thanks to all that helped me over the weekend.

HackIT, 2017-05-05:
I guess to make it work with your mac you have to add some rules somewhere... a like we do for linux with udev.

resqmedic, 2017-05-05:
If ya figure it out lemme know, I'd rather use the mac

rowens83, 2017-07-07:
Im having a similar problem trying to flash betaflight to dys elf f3 controller.  when trying to flash i get no repsonse from bootloader error.  Ive tried every combination with full chip erase, no boot sequence , manual baud rate.  all combination with and without using boot button.  Any help would be greatly appreiciated.

mikeller, 2017-07-07:
@rowens83: Have you looked into https://github.com/betaflight/betaflight/issues/3437#issuecomment-313197226 as a potential source for your problem?

sorad10, 2017-08-28:
HI.
Just bought a Betaflight F3 FC, it is not detected from my computer, does not power up, no sound no lights. When I connect the battery, it does indeed power up. So far, I soldered the ESC pwr & ctls. Am I missing anything? (Yes I have a MAC and I was trying to run Betaflight on my Windows 10 partition)

Any help is appreciated.

Thanks

mikeller, 2017-08-28:
@sorad10: GitHub is for question and discussions around *development* of Betaflight. For questions about hardware problems, please contact your supplier, or (RCGroups)[https://www.rcgroups.com/forums/search.php?searchid=62152622&query=betaflightf3].

Also, if there are no lights when you are plugging your flight controller into USB, it sounds a lot like the USB port on the board is broken.

sorad10, 2017-08-28:
OK will do.

Ub6ib9, 2017-09-28:
I'm having the same problem with my betaflight f3 board. Goes into dfu mode but won't flash. Says "detected device with total flash size 256kib" and won't flash. And I'm using windows....

Ub6ib9, 2017-09-28:
It's bad enough that 2 of my QWAD's were out of commission due to betaflight tweekin out on the failsafe tab and receiver tab going crazy deleted numbers.... After a month I finally found that....  And flying again after replacing values in the min and Max on Failsafe tab and min mid and max in receiver tab. Now I just want to update the firmware on my BFF3 board.... And cant

jflyper, 2017-09-28:
@Ub6ib9 Yours also sounds like a support request. GitHub is for question and discussions around *development* of Betaflight. For questions about hardware problems, please contact your supplier, or [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62152622&query=betaflightf3).

closed 2017-09-28 by jflyper

Ub6ib9, 2017-09-28:
It's because you guys keep developing and developing but cause problems by going too far. My board has no hardware issues. It worked fine the last time I flashed it and only flown a few times. Time to switch to racefight....

wind0r, 2017-09-28:
DFU is done by STM itself. Nothing Betaflight can change there or ever changed. There isnt any way for a Firmware to change how DFU works. There isnt any way Betaflight can change it so it works better or worse. This just can be changed on Computer / Enduser side and not with any code in Betaflight.

So since you said there isnt any hardware issue there just can be something strange with your Computer.

Please remember this is a Issue tracker for the hobbiest that develop Betaflight further. This is not a Support Tracker for flashing and configuration Issues. As @jflyper said consult your supllier, Facebook and/or RCGroups.

This Issue tracker is not for generating the best flashing experience and fixing each and every operating system driver issue. Since every flashing Issue i know just is some driver issue and i personaly never had any issue on any os to flash my boards this isnt something betaflight can fix for you. Even Switching the Firmware wouldnt help you much since those are also using the same ways and tools to flash your device as i said DFU is done by STM.

Ub6ib9, 2017-09-28:
That's the thing I don't have a problem getting into dfu mode. And I have
flashed to other play controllers with the same computer without a problem.
Ever since betaflight 3.2 came out there has been issues. Like the
resetting of the values in Fail-Safe and on the receiver tabs. Which the
two boards I was able to flash don't have that problem anymore. Which is
why I want to update my betaflightf3 board so I don't have the same problem
every time I plug into betaflight. And my betaflightf3 board goes into dfu
mode but won't flash. The drivers are fine and the computer flashed two
other boards flawlessly. And I have been dealing with this for over a month
where else would I turn. I have tried RC groups and everything else. I'm
not trying to be a dick it's just something out of my control I have been
flying quads since baseflight.... I don't know how I couldn't do anything
different. And the last board I figured I would have problems on betaflight
with is my betaflightf3 board.

jflyper, 2017-09-28:
This is the BFF3 thread.

https://www.rcgroups.com/forums/showthread.php?2795213-Betaflight-F3-Flight-Controller-OSD-PDB-SD-card-BEC-current-sensor

Ub6ib9, 2017-09-28:
> This is the BFF3 thread.
>
> https://www.rcgroups.com/forums/showthread.php?2795213-
> Betaflight-F3-Flight-Controller-OSD-PDB-SD-card-BEC-current-sensor

Thank you for the information. I love betaflight! I just hate when things
quit working, especially when I didn't do anything for it to trip up.
Hopefully I can figure it out.

Thanks again
Kyle