betaflight/betaflight #2982: Motors overheating tried everything please help

Dmulg, 2017-04-29:
Hi,
I reccently finished a build, and I am having a problem with my motors getting very hot after ~45 seconds of flying. I have tuned and retuned, but nothing helps. Any advice? ...I think the filters could help but I am unsure what to change. This is my 5th build and I have never had this issue before. 

Quad(FC and motors soft mounted):
QAV-R 5"
Tmotor F40 2400k/v
Rotorgeeks SSD
Spedix 30a escs (Dshot600)
Hq V1S 5040 tris
1300 4s

nathantsoi, 2017-04-29:
what version of betaflight? did you try turning your pids way down to
start? probably start with p. also, feel free to post a blackbox log

HackIT, 2017-04-29:
Hey,
I bet your quad is very noisy and your FC too much filtered which makes motors to warm up very hot...
 you should take a look at it : https://github.com/betaflight/betaflight/wiki/Gyro-&-Dterm-filtering-recommendations
You can also try using another esc protocol... and be sure that your escs are capable to run the choosen protocol... Take a look at your propeller when they spins IDLE to notice if they are unbalanced(which makes noises)...
Softmount your FC and maybe your motors with some 3dprinted tpu pads to reduce a bit more the noise...
If your quad frame uses separate arms softmount them too with some double face foam tape.
Be sure your FC is well configured(esc protocol/calibration, failsafe, flight mode, rx, filters and pids)
At this moment you'll be okay for a glitch free flight.

AleksanderAzizi, 2017-05-13:
I have the same issue, but my motors overheat after just 5-10 seconds of hovering.
Don't know what caused the issue, as it worked fine a couple of weeks ago.

_Martian II 230mm.
Colibri Race V2 FC (bf v3.1.7).
Returner R4 2206 2600kv.
KISS 24a RE ESC (Dshot600).
HQ V1S 5x4x3.
1400mAh 4s._

Dmulg, 2017-05-13:
Hey,
What ended up helping on mine was to disable the notch filters, and tune the D gain way down (10-15 range)... I also put the yaw filter at 80... not sure how much that effected it but it flys well now and the motors come down cool.

spuder, 2017-07-20:
@Dmulg Do you mean that you reenabled notch filters? Wouldn't disabling notch filters make your motors hotter?

closed 2018-04-21 by DieHertz