betaflight/betaflight #2988: Full throttle loss of control

Wtcheesecakes, 2017-04-30:
When giving full throttle I lose control of the craft.
Betaflight 3.1.7 spracing f3 and

netzwergracer, 2017-04-30:
Must be a problem of your copter though.
No issues with full throttle on 3 copters on Betaflight 3.1.7 here.

Wtcheesecakes, 2017-04-30:
Hmm should I try black box it? See if it's ESC or what tf

jflyper, 2017-04-30:
@Wtcheesecakes You can at least provide what "loss of control" is like.

Wtcheesecakes, 2017-04-30:
@jflyper hi there its when i go full throttle it almost pushes the craft in one direction almost as if it is forcing it into a nose dive and i cant add any roll or pitch input when its in that state

RipperDrone, 2017-04-30:
Is airmode = On and I terms are high enough to maintain attitude while punching?

RipperDrone, 2017-04-30:
Maybe this references to https://github.com/betaflight/betaflight/issues/2952

Wtcheesecakes, 2017-05-01:
EDIT: i have flown once more and it seems i dont loose control it just tilts backwards or forwards on punch out

RipperDrone, 2017-05-03:
then u need to increase pitch I to keep pitch attitude

MrDubs, 2017-05-05:
I'm also running an spracing f3 in acro with airmode on and having similar issues. On a full throttle punch out the quad pitches back I've tried raising pitch I to the point were I get oscillations but it makes no difference. Strange thing is I was previously running cleanflight v1.14.2 on multiwii and it flew perfectly with none of this behaviour. Wondering if it's something to do with the filter settings as I know my quad is quite noisy but never really played with these before.

Wtcheesecakes, 2017-05-05:
@mrdubs I suggest trying anti-gravity on the same switch as your arm and see if that helps :)

MrDubs, 2017-05-05:
Will try that. Now off to have a play so will let you know how I get on.

MrDubs, 2017-05-05:
For an update unfortunately anti gravity made no difference to the issue. Had a mess around with the gyro and dterm low pass filters and that also dint help the problem. Don't get me wrong the change in attitude is probably only a few degrees but is definitely there and it weren't with cleanflight. Maybe I'm just s*#t at tuning pids lol

Wtcheesecakes, 2017-05-06:
you can turn up the antigravity amount
https://www.youtube.com/watch?v=SmSWZFjXBGM&t=412s&ab_channel=JoshuaBardwell

MrDubs, 2017-05-08:
Yer just watched the video ide herd of anti gravity but never knew what is was all about sounds like its gonna be the answer to my problems will give it a try on a stronger setting at some point this week and let ya know thanks for your help.

martinbudden, 2017-08-17:
Duplicate of https://github.com/betaflight/betaflight/issues/3016

closed 2017-08-17 by martinbudden