betaflight/betaflight #2989: TBS Smart Audio Omnibus F4 Pro V2/v3 Not compatible

jaklosrc, 2017-04-30:
I have had a working setup bf 3.1.7 with a v1 Tbs unify pro hv it only worked on uart 3tx not uart6 tx. Long story short antenna ripped off and the uvl connector is trashed.  No problem I have a spare brand new same batch lot number tbs unify pro hv v1 batch 1003.  Plug it in and well lets say i no longer have access to it via the vtx sa menu.  it is all dashed out even after going in the statx menu to enable smart audio.  now if i go into the vtx tr tramp menu i see what i normally would have under the vtx sa menu band channel power set i can adjust the settings in the menu but it will not save or switch the channel.  after beating myself up over this i opened a brand new omnibus f4 pro v3 installed 3.1.7 setup ports for uart 6 for a serial rx and setup uart 3 tx for smart audio.  to no avail the unify pro hv no longer speaks to the omnibus the one that's missing the antenna nor a brand new one.  nor on a brand new omnibus.  i have gone thru and tried firmware 3.1.1-3.1.7 with no working progress.  My problem is it was once working and just quit working that is why i tried a brand new one and behaviour is the same.

jflyper, 2017-05-01:
@jaklosrc There are very useful comments in the RCG thread that you have cross posted.

jflyper, 2017-05-01:
I think this can be closed. It's a pure support request.

kaspergrubbe, 2017-05-02:
@jflyper Which RcGroups thread are you referring to?

jflyper, 2017-05-02:
@kaspergrubbe 
https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro/page107#post37416144

closed 2017-08-03 by jflyper