betaflight/betaflight #3008: X-racer F303 SPI servo tilt not working when you remap outputs

CuriousMindZA, 2017-05-03:
My setup - esc's on motor 1,2,3,4 gimbal servo on motor 5 output, sbus rx and uart2 drives osd.
If you enable the servo tilt function motors is shifted( but I think they should not in 3.1.7) and if you then remap motor resources to their original pins (motor1 to 4 output for me) the servo output pin does not work any more. Setting motor resources to none in cli and servo 1 to motor output 5 the servo works again. If you then remap motor 1 to its original pin all 4 motors reappear even though  2,3, and 4 should be unassigned. Also if you enable servo tilt and remap motors only two actually run but a resource list show all should work. if you disable servo tilt all work as expected.

jflyper, 2017-05-03:
@CuriousMindZA 
I didn't try to understand what you have explained (too much text ... :sob:), but one thing you should know is that BF3.1 and later has changed the way SERVO_TILT work.

It does NOT shift outputs anymore, but uses two servos from configured servos. More specifically, in your case, configuring additional servos for gimbal has nothing to do with motor outputs.

Below is an example of how servos (including gimbal servos) work on TRI + gimbal + 2 aux pass through case. Please examine this drawing, and if you still have something unsolved, please paste relevant part of the outputs from `resource` and `resource list` commands.

[image]

CuriousMindZA, 2017-05-03:
Short and sweet :)  - Enable servo tilt = motors DO shift and remapping does not work. Using 3.1.7 reflashed fc 2 times.
[image]

jflyper, 2017-05-03:
You haven't read what I've wrote 😭 

Try this.

1. Connect your pitch servo to motor output 6, and roll servo to motor output 7.

2. Use following CLI commands to map motor outputs 6 & 7 to logical servo 1 & 2.
```
resource servo 1 b9 # PWM6
resource servo 2 a2 # PWM7
```

CuriousMindZA, 2017-05-03:
Thx. But for what its worth. I just reflashed FC with x_racerspi version 3.1.7. Configured ports and rx(sbus). Then I went to motors tab and tested motors. All 4 is running. I then enabled servo tilt and save. I go back into motor tab and repeat test. Only 2 motors run. I will make a vid of what I do and what happens. This without any remapping just enabling servo tilt. Thanks for your time. :)

jflyper, 2017-05-03:
@CuriousMindZA 
I suspect you still have servo 1 mapped to PB8 which is preventing from motors 3 and 4 working (they all share the timer TIM4).

CuriousMindZA, 2017-05-04:
Never Mind. I flashed Cleanflight 2.0.2 on my FC and now it works like it should. Same hardware config as above.

jflyper, 2017-05-05:
Well, that's weird, because CF2.0.2 is essentially identical to BF3.1.7.

But you can close this issue now if you are satisfied.

johnty, 2017-05-05:
sorry to piggyback on this, but i believe i'm facing a similar confusion in terms of resource mapping, in the context of building a brushed bicopter on a naze.

based on what i'm understanding/experiencing, smix servo slots correspond to the motors tab page. i can verify this by seeing that a smix of:

```
smix 0 4 2 100 0 0 100 0
smix 1 4 1 100 0 0 100 0
smix 2 5 2 100 0 0 100 0
smix 3 5 1 -100 0 0 100 0
```

shows yaw and pitch contribution to the servos 4 and 5 in the servos tab. If I change those to 0/1 instead of 4/5, i see the first two servos respond instead. so far so good.

however, my actual servos are connected to physical pins 1 and 2 on the MCU:

```
# resource
resource BEEPER 1 A12
resource MOTOR 1 B06 #pin 3
resource MOTOR 2 B07 #pin 4
resource SERVO 1 A08 #pin 1
resource SERVO 2 A11 #pin 2
```

so there's another layer of mapping that assigns the servos at the resource (pin) level to their associated servo "slot" value - as shown by the mappings that occur beneath "servo slots" take place. (e.g. green line: mixer defined channels, and the flow from the two gimbal channels to logical servos 2/3, and downwards)...

in terms of the specific challenge i'm facing, the above settings, if I load the "BI(copter" configuration, I get correct servo action. it's clear that "servo 1" in terms of resource list gets assigned to 4 (smix)/5(servos tab). However, if I'm using CUSTOMAIRPLANE, where in theory servos are enabled, there is no servo movement. the explanation i can think of is that for this profile, this mapping does not exist. while I'm able to work around this by hardcoding the BI profile and recompiling, a better way (especially for tweaking) would be to be able to effectively use the CUSTOMAIRPLANE mixer via configuration, if at all possible).

and am i also correct in understanding that for a preset (non-custom) profile (e.g. loading mixer BI, followed by mmix BI and smix BI), the actual mix is fixed - as after loading, adjusting the smix doesn't seem to do anything except display different values?

jflyper, 2017-05-05:
> However, if I'm using CUSTOMAIRPLANE, where in theory servos are enabled, there is no servo movement. the explanation i can think of is that for this profile, this additional layer of mapping does not exist by default.

Is this your main problem statement?

johnty, 2017-05-05:
yes, it is my main problem. i think it is due to my lack of understanding of how an extra layer of mapping exists (or doesn't).

> I think CUSTOM_AIRPLANE enables servo, but there's no default servo mix (naturally), thus no servo movement...

when you say servo mix, its not simply a "smix" setting right? because adding smix simply gives the desired input (from FC, stick, etc), to the values in the servos tab (and this part works fine). but it doesn't actually map it to the physical servo attached to the output pin. so the question can be boiled down to: 

* How do I add a servo mix to the CUSTOM_AIRPLANE mixer configuration? Based on my understanding so far, the steps of of setting up resource mapping, loading mixer profile, and adding smix is missing one crucial component.

(and just so its not a fun mental exercise: the reason for trying to do this is the fixed presets don't allow you to adjust smix; so CUSTOM_AIRPLANE becomes useful in many cases. otherwise, i have to resort to recompiling and re-flashing when tweaking the mixer behaviour).

let me know if this made the question clearer... thanks for your patience!

jflyper, 2017-05-05:
@johnty 
Did you try the "Custom Airplane with Differential Thrust" example?
By loading custom mmix and smix, you will be able to control two motors, and observe four servo (3 through 5) control signals on servo slots.

From there on, you would merely assign these logical outputs to physical pins, and your
```
resource MOTOR 1 B06 #pin 3
resource MOTOR 2 B07 #pin 4
resource SERVO 1 A08 #pin 1
resource SERVO 2 A11 #pin 2
```
Look fine (servo 3 and 4 are missing, but suffice for a tutorial purpose).

johnty, 2017-05-05:
@jflyper 

yes, the smix stuff to servo slot mapping appears fine - stick input or stabilization control will cause the right servo channel to respond as shown in the servos/motors page in configurator.

but when you do the resource mapping, those servo assignments to hardware pins are not mapped to the servo slots above. (e.g. in the example above, we assigned servo "1" to A08 or pin 1, but is there an implicit assumption that this should fit into servo "slot" 1 of the smix servo index?  if we do make this assumption, then i can report that it's not actually working as intended right now.

if someone can point me to the code where this level of mapping occurs, that could be helpful as well. (my current understanding of what goes in servos.c/h is that its all happening on the smix level, but the resource mapping occurs elsewhere).

jflyper, 2017-05-05:
(1) Servo slots are mapped to logical servos.
Those defined in smix comes first, gimbal next, then channel forwarding.
For the "CADT - Custom Airplane with Differential Thrust" example, servo slots 3,4,5 and 6 are used, so these are mapped to logical servos 1, 2, 3 and 4.

(2) logical servos are mapped to pins using resource command.

I will do the actual output test with CADT if this is not the case.

johnty, 2017-05-05:
>(2) logical servos are mapped to pins using resource command.
great! this part makes sense.

>(1) Servo slots are mapped to logical servos.

i think this is still where its a bit murky. in this case, for the following:

>... servo slots 3,4,5 and 6 are used, so these are mapped to logical servos 1, 2, 3 and 4.

where does the slot 3, 4, 5, 6 -> logical  servo 1, 2, 3, 4 assignment take place? is this a hard coded per-mixer configuration setting somewhere, or can you adjust this on the fly somehow?

and based on experimentation so far, THIS final mapping is whats not actually working for me on the CADT mixer example using CUSTOMAIRPLANE. i do know it physically works for my hardware because the BICOPTER profile works, with servo slots 5 and 6 (smix index 4, 5) working with logical servos 1 and 2. if i moved a logical servo to replace one of the pins, e.g. have servo 3 on the pin for servo 2, then it stops working as expected. so there's an explicit mapping somewhere here...

at this point i've obtained what i need for my [bicopter project](https://www.youtube.com/watch?v=lL0-NaDt3-g) by hardcoding the correct smix into the BI profile (just need a lot of tuning! haha), but it would be nice to fix a potential bug if it exists (and to generally get a better understanding of things!)

thanks again,

jflyper, 2017-05-06:
> where does the slot 3, 4, 5, 6 -> logical servo 1, 2, 3, 4 assignment take place? is this a hard coded per-mixer configuration setting somewhere, or can you adjust this on the fly somehow?

Take a look at the `writeServos` in `src/main/flight/servos.c`.

johnty, 2017-05-17:
hmm, just came back to here after someone commented on a youtube video and was going to refer them to this page... i thought i had posted a reply but guess didn't hit the "comment" button... anyway, i got it to work and it behaves correctly. what got me confused was the fixed nature of the servo slots mapping layer for a given mixer configuration. e.g. fix slots 1 and 2 for gimbal, and then slots 3 onwards become assignable for smix. looking at writeServos made this point more obvious :)

so my current working configuration is for CUSTOM_AIRPLANE:

```
#resource
...
resource SERVO 1 A08 #physical pin 1 on naze
resource SERVO 2 A11 #physical pin 2 on naze
...
# smix
smix 0 2 2 100 0 0 100 0    #servo slot 3
smix 1 2 1 -100 0 0 100 0  #servo slot 3
smix 2 3 2 100 0 0 100 0   #servo slot 4
smix 3 3 1 100 0 0 100 0  #servo slot 4

```

so resource SERVO 1 (logical) falls into servo slot 3 (2). this cannot be changed using any configurable mixer CLI command, but is set by the profile.

in the bicopter mixer profile, resource SERVO 1 falls into slot 5(4) instead, so for smix it should be shifted by two.

everything makes perfect sense now. thanks!

jflyper, 2017-05-18:
Great to hear!!!

Have fun 😃

closed 2017-10-21 by jflyper