betaflight/betaflight #3011: bicopter mixer profile, pitch for left motor reversed; no throttle response

johnty, 2017-05-03:
i'm building a brushed bicopter using a naze32 board and have managed to bridge the PWM outputs on the board to drive the two servos. some things i've noticed:

- first problem: by default the pitch contribution to the LEFT servo appears to be reversed. after changing [this line](https://github.com/johnty/betaflight/blob/master/src/main/flight/servos.c#L121) to

`{ SERVO_BICOPTER_LEFT, INPUT_STABILIZED_PITCH, 100, 0, 0, 100, 0 },`

makes both servos tilt the same (i hope correct?) way when pitching forwards. (I could have it backwards, in which case the RIGHT servo should be flipped, but this is minor...)

- second problem: despite what looks like a correct mmix:

```
# mixer
Mixer: BI

# mmix
mmix 0  1.000  1.000  0.000  0.000
mmix 1  1.000 -1.000  0.000  0.000
```

I get no throttle response at all. The motors are wired up fine: motors tab works; arming spins them up as well.

Any idea where I should be looking?

This is what it looked like [last night](https://www.youtube.com/watch?v=0sBkS7hgmEU) on CF1.x, where I was able to get throttle but wrong servo mix :)

Thanks!

johnty, 2017-05-03:
so, some TL;DR points of discussion:

1. may be more consistent to apply the USE_UNCOMMON_MIXERS check on both smix and mmix settings?
2. is there actually a bug for custom smix/mmix's, or did i not use them correctly? based on my experimentation i couldn't find any profile (including CUSTOMAIRPLANE, etc which should have enabled servos) that allows arbitrary mmix and smix's to be applied correctly.
3. in terms of the actual bicopter preset, i think the flipped pitch may be good to have since the most obvious (default?) way would be to mount servos with horns facing outwards, especially considering yaw is already making this assumption and operating correctly.

mikeller, 2017-05-29:
Point 1 makes good sense, I think that is an oversight from when USE_UNCOMMON_MIXERS were implemented. Also, note that non-quad support and testing is very limited in Betaflight, so 'there be dragons'. ;-)

johnty, 2018-06-03:
sorry its been a while, but as far as i can tell these changes have been merged in. 1 and 3 seem to be in the current code, and 2.), if i can recall, was actually due to my misunderstanding of how the logical servo slots worked. so probably good to close this :)

thanks for checking up on this!

mikeller, 2018-06-03:
All good, just cleaning up. ;-)

closed 2018-06-03 by mikeller