betaflight/betaflight #3013: X quad (1,2,3,4) config not working on CC3D?

willback510, 2017-05-04:
Just updated my cc3d to Betaflight coming from libre pilot. when I was setting it up, I noticed that if I put the quad configuration in the 1,2,3,4 layout, it sort of wiped out the controller. every time this happens I have to use libre pilot to restore the fc to a point where the pc can recognize it again, not sure why this is a problem. I have a spectrum sat receiver on the mainport, may that be an issue? The normal X config worked fine, the reason I tried the 1,2,3,4 config is because I was having some channel mapping issues, and I was hoping to see if this could solve it. help pls. I want to fly

jflyper, 2017-05-04:
@willback510 
Is it possible to configure it as QUAD1234 immediately after the full chip erase flash and reboot then come backup and connect to the configurator?

AndersHoglund, 2017-05-04:
I have no problem configuring QadX1234 on CC3D with BF3.1.7. Flashed with "Full Chip Erase" and only "QuadX1234" selected. Using target "CC3D", flashed over UART1/Main port. Are you using target "CC3D_OPBL" ? 

    # diff
    # Betaflight / CC3D 3.1.7 Apr  3 2017 / 21:54:46 (e1c4b5c)
    mixer QUADX1234
    
    profile 0
    
    rateprofile 0
    rateprofile 0
    
    
    #

willback510, 2017-05-04:
All I know is that when I select the quad x1234 config and hit save, it disconnects the controller,, the controller restarts,  and the status led slowly fades in and out. I believe this is an indicator that it is in bootloader mode? IDK. I'm still kind of new to this whole fc setup deal so yea.

AndersHoglund, 2017-05-04:
OK, sounds as you still have the OP bootloader onboard and using the betaflight CC3D_OPBL target file. 
Slowly fading status LED indicates the OP Bootloader is running. (The primary STM boot loader has no such indicator). So, I tried to duplicate by first restoring the OP boot loader and then via GCS flash CC3D_OPBL. 
And then I selected "QuadX1234". No problem after save and reboot, even though reboot takes a while. I have nothing connected to the FC, i.e. no Spektrum Sat etc etc. 

    # diff
    # Betaflight / CC3D_OPBL 3.1.7 Apr  3 2017 / 21:55:18 (e1c4b5c)
    mixer QUADX1234
    
    profile 0
    
    rateprofile 0
    rateprofile 0
    
    
    #   

Have you tried disconnecting the Rx on UART1/Main port? It may cause strange things as at least the STM boot loader listens on UART1. Do not know if the OP boot loader also does this. Normally you would have a sat Rx on UART3, the Flexport. Try moving the sat there. 

And also, do your self a favour and get a USB serial adapter (FTDI, CP2103 etc etc) and flash the CC3D target via main port using BFC flasher, thus erasing the OP Bootloader and leave the OP/LP world behind. Only downside is that you can NOT flash via the USB-VCP port in the future, only via Main port. But you can always restore the OP boot loader later on, if you fell the need.

AndersHoglund, 2017-05-04:
Did some more tests on CC3D_OPBL, now with a LemonRx diversity sat connected on Main. Still possible to select QuadX1234. However, I did notice some times after a save and reboot USB did not not register properly with BFC, needed to power cycle the FC. I think this is a problem with my particular FC, the USB connector is becoming worn out..... 

    # diff
    # Betaflight / CC3D_OPBL 3.1.7 Apr  3 2017 / 21:55:18 (e1c4b5c)
    mixer QUADX1234
    
    feature -RX_PPM
    feature RX_SERIAL
    map TAER1234
    serial 0 64 115200 57600 0 115200
    set serialrx_provider = SPEK2048
    profile 0
    
    rateprofile 0
    rateprofile 0
    
    
    # 

Tests done with (1) Rx unpowered, (2) powered but not connected to Tx and (3) powered plus connected to Tx. So I think this rules out any UART1 related problems.

willback510, 2017-05-04:
Ok so I reflashed a fresh copy of bf on it via librepilot, then I went to the bf configurator, set it to quadx1234, save and reboot, now I have power but no status light at all. I did a few power cycles, nothing

AndersHoglund, 2017-05-05:
Not sure what is different with your setup compared to my tests... Have you tried changing any other setting besides the mixer "QuadX1234"? I.e. is the problem related to this particular parameter or is it a general problem with reboot after saving any parameter?

AndersHoglund, 2017-05-07:
Ahh, sorry I am slow... There is a well known issue with CC3D_OPBL and Windows when starting/restarting. You MUST power up the FC without USB connected and let betaflight do all initialization. Then you can connect USB and Windows will find the STM USB-VCP device. If USB is left connected, Windows will see the OP BooLoader USB device, and lock on to it. Windows will NOT re-select the STM VCP-USB device as it becomes available when betaflight is up and running. This is most probably the major reason for your struggling, and yet another strong reason to erase the OP boot loader by using the CC3D target.

willback510, 2017-05-07:
That did it! Thanks to everyone who responded, really liking the community so far!

closed 2017-05-07 by AndersHoglund