betaflight/betaflight #3016: I term windup at full throttle when airmode is off

totterfree, 2017-05-04:
Both I and a friend have seen an issue on 5" builds which do not have airmode on. This issue spawned during one of the recent 3.1.x versions. It seems that when when you are near or at full throttle, the I term seems to wind up, making the quad uncontrollable. Yawing, pitching, and rolling seem to do nothing until a certain point, at which the quad seems to break loose. It then flips 180 degrees, (usually bringing it to a stop, or causing a crash) and continues flying ok until the next full or high throttle punch out. Turning on airmode and antigravity mode seem to cause the issue to go away

Steps to reproduce:
1. Reset to default bf settings
2. Configure RX settings
3. set arming to a switch 
4. DO NOT enable airmode or antigravity mode
5. While flying, do an extended full throttle punch out
   - once the craft is up to speed, attempt to make a slow, smooth 180 turn
   - you will feel as though the quad has yanked control from you and pulled a sharp aggressive turn without your stick input

Steps to remedy
1. turn on airmode
2. turn on antigravity mode
   - I haven't yet tested which one of these fixes the problem
   - however, doing both definitely causes the issue to go away
3. the craft should now respond as expected

I have identical builds all using the same frame, fc, motors, and ESCs. Only one of them displayed this problem. After checking the bf settings, I realized that the build showing the problem was the only build which had airmode turned off. 
I also had a much lighter one-off build which seemed to display this issue. After turning on airmode and antigravity this build also flew exceptionally better.

RipperDrone, 2017-05-04:
Wow, seems you are the first to have systematically reproduced this glitch which has been reported 10 times+ on various builds. Really hope that it now gets crystal clear for the core team how to fix it. This issue's story seems to get spread around so many people (e.g. in the BF FB group) start complaining about 3.1.7.

It seems majority of cases reported can be fixed by reverting back to 3.1.6 so it seems to be specific to the actual firmware only...

martinbudden, 2017-05-05:
@borisbstyle , probably unrelated, but I noticed an error in:
https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_rc.c#L169

Presumably the multiplier should be `0.001f` not `0.0001f`.

McGiverGim, 2017-05-06:
I don't know if mine is the same problem, but if it can give more data to help solve it: when I do a throttle punch in acro mode the quad does not go up straight, it starts to roll to the right and pitch to the front. If I enable air mode, the quad then starts to maintain attitude without problem, and the throttle punch goes perfectly straight.
In the blackbox I can see the motor 2 at 100% but it seems not to be sufficient.
Regards!

digifaps, 2017-05-08:
I have exactly the same behaviour as posted above.

marcroe, 2017-05-08:
I can also confirm this. I believe enabling Airmode fixes it (not anti-gravity). 
How I reproduced it: A big power-loop at full throttle, i.e., don't cut throttle early, start pitching back when still at full throttle.
Also, see issue https://github.com/betaflight/betaflight/issues/2988 which could be a duplicate or related at least.

jameth, 2017-05-11:
This is absolutely on point.

With Airmode off, I am almost guaranteed to have this happen on swooping moves with a punch out, quad randomly pitches and rolls 

With Airmode on, perfect punch outs at all times, problem disappears.

Have tanked my micro multiple times because of this doh!

TazRC, 2017-05-16:
Same behavior here on 4" quad with Air set up but turned off. Antigravity not set up.

digifaps, 2017-05-16:
same here , have a second quad , 4 inch too , exactly the same behaviour.

laprade65, 2017-05-16:
I am seeing a similar issue with throttle chop but also if I try a backflip it tumbles and never regains control.  I should have logs I can look through this afternoon

borisbstyle, 2017-05-16:
Next version will have airmode active on full throttle when airmode disabled.

That will solve the issue

rodgerramjet, 2017-05-29:
I kinda like it... it reminds me to turn on airmode :)

martinbudden, 2017-08-01:
@borisbstyle , has this been fixed?

Saxin, 2017-08-03:
@martinbudden this one I think - 41d8455b5336f7141e63c6c670b48ba0f9366c12

martinbudden, 2017-08-17:
Fixed by commit https://github.com/betaflight/betaflight/commit/41d8455b5336f7141e63c6c670b48ba0f9366c12 (no PR associated).

closed 2017-08-17 by martinbudden