betaflight/betaflight #3022: Motors Tab Issue on v3.2

kc10kevin, 2017-05-05:
Hi all,

Couldn't find any issue brought up on this, so thought I'd do it.

On v3.2 firmware, in the motors tab (BF Config 3.1.1), it is not possible to get the motors sliders to activate.  v3.1.7 works fine.  Something changing in v3.2 for it not to work?   Tried on several targets:  FuryF3, FuryF4, Revo, OmnibusF4SD, DoDo.  Am i missing something?

Here is a video of the behavior:  https://www.youtube.com/watch?v=lyumXYSS3yA

First test is with v3.1.7, then flashing v3.2.  Enable motors slider is on after flash and not able to use sliders.

Thanks in advance!

MJ666, 2017-05-05:
Same here for the latest's build. BTW the CF configurator still works.

There are some more issue I have seen during testing the AlienFlight F7:

- The VBAT and CURRENT_METER at the configuration page does not work properly anymore since the feature are removed from the code.
- The servo tab is not working

kc10kevin, 2017-05-05:
I found the same issue with the Current and VBat with the boards I have.   Any idea how they will be configured in the future?

nathantsoi, 2017-05-06:
i see the same thing:

[image]

how are we suppose to debug this? there is no indication of what is broken with the same error code `return MSP_RESULT_ERROR` in every msp message.

i suspect this is due to one of your recent changes @hydra, if you could plz take a look

nathantsoi, 2017-05-06:
hum... i see... i think it's actually here: 5d602d69ff96ab892f47438f275cd6f3a506046c we do get the code. it's just been removed :)

nathantsoi, 2017-05-06:
gaaaaahhh, can no commit's be reverted!? any suggestions @hydra ?

TheAngularity, 2017-05-06:
@kc10kevin  @MJ666  @nathantsoi 
all this is fixed here https://github.com/betaflight/betaflight-configurator/pull/488

hydra, 2017-05-06:
The solution is to update the configurator, not revert commits.  The other guys are on it.  I've not had time to help out much recently.

nathantsoi, 2017-05-07:
nice work @mikeller, just tried this out and it appears to be fixed in https://github.com/betaflight/betaflight-configurator/pull/488

closed 2017-05-07 by nathantsoi