betaflight/betaflight #3029: F411 TIM2 (LED STRIP) hard faulting?

jflyper, 2017-05-06:
Kind of a reminder.

This happens to REVONANO target (F411), trying to activate the LED strip assigned to TIM2.
As soon as LED_STRIP feature is enabled and rebooted, FC wedges during initialization.
Some SWDing revealed TIM2 interrupting as soon as nvic is called for vector initialization (which doesn't happen with other timers?).

```(gdb) cont
Continuing.
^C
Program received signal SIGINT, Interrupt.
TIM2_IRQHandler () at ./src/main/drivers/timer.c:646
warning: Source file is more recent than executable.
646    _TIM_IRQ_HANDLER(TIM2_IRQHandler, 2);
(gdb) where
#0  TIM2_IRQHandler () at ./src/main/drivers/timer.c:646
#1  <signal handler called>
#2  NVIC_Init (NVIC_InitStruct=<synthetic pointer>)
    at ./lib/main/STM32F4xx_StdPeriph_Driver/src/misc.c:157
#3  dmaSetHandler (identifier=<optimized out>, callback=<optimized out>, 
    priority=<optimized out>, userParam=<optimized out>)
    at ./src/main/drivers/dma_stm32f4xx.c:90
#4  0x0800dd92 in ws2811LedStripHardwareInit (ioTag=<optimized out>)
    at ./src/main/drivers/light_ws2811strip_stdperiph.c:123
#5  0x81000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

mikeller, 2017-08-13:
@jflyper: Has this been resolved with the timer fixes for F411?

jflyper, 2017-08-13:
@mikeller This one dropped out of my mind. Let me check it out.

jflyper, 2017-08-13:
Works like a charm, but no theory behind it ... will close this for now 😅

closed 2017-08-13 by jflyper