betaflight/betaflight #3033: SPRACINGF3 (ACRO) - DSHOT600 + FAULTY PIN REASSIGNMENT- DMA issue.

778998, 2017-05-06:
Hi all, hope  you guys can help.  Im using the following in my setup.

SPRACINGF3 ACRO Flight Controller.
DYS XS30A (with signal cap removed).

All motors work perfectly on Multishot but only 3 on DSHOT600.
Now, with having said that, I have 2 faulty pin headers (pin 4 and 5). When i was on multishot i had the resource assigned to pin6 and that worked perfectly fine.  But now that im trying to run dshot600 i have an issue where motor 1-3 will work but not motor 4.

I have read on the DSHOT page that you have to reassign resources but then again that was for the MINI and EVO boards, mine is just the normal Acro board.  In the DMA section below my Motor 4 does not appear and is most likely my issue.  Can anyone please shed some light on this please ?

`CLI

# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 B09  (has been set to pin6[B09] as pin 4 & 5 are faulty)
resource MOTOR 5 B08  (has been set to none but im listing it here just incase)
resource MOTOR 7 A02  (has been set to none but im listing it here just incase)
resource MOTOR 8 A03  (has been set to none but im listing it here just incase)
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00

# resource list
IO
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: ADC_BATT 
A05: ADC_CURR 
A06: MOTOR 1
A07: MOTOR 2
A08: FREE 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: FREE 
A13: FREE 
A14: SERIAL_TX 2
A15: FREE 
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: LED 1
B04: FREE 
B05: FREE 
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE 
B09: MOTOR 4
B10: FREE 
B11: SERIAL_RX 3
B12: FLASH_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: BEEPER 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: FREE
DMA1 Channel 3: MOTOR 1
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

# `

mikeller, 2017-05-07:
```
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: FREE
DMA1 Channel 3: MOTOR 1
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

That doesn't look good. The TIM4_CH4 that is used for motor 6 / PB9 does not have any DMA channels available, and you need one of those to run DShot.

You could try using https://github.com/betaflight/betaflight/wiki/reference/stm/stm32f3_pins_timers_dma.pdf to figure out a timer / DMA assignments that gives you one DMA channel for each of motors 1, 2, 3, 6, and then build your own custom target.

778998, 2017-05-07:
Thanks Mikeller, I decided to move to pin 7, updated the resources, and now my motor 4 is being detected on DMA 1 Channel 5. :)

mikeller, 2017-05-08:
Good stuff. Can you please close this ticket?

closed 2017-05-13 by 778998