betaflight/betaflight #3036: Mixer Dropdown items Heli 120 and Heli 90 do nothing

hursto75, 2017-05-06:
These mixer Drop downs do not apply a mix of any kind.  If they are not used remove them form the drop down.

jflyper, 2017-06-07:
Yes, they should be removed, among with other non-working mixers.
I think it is easier to remove them from BFC, then from BF.
Can you submit an issue PR on BFC repo?

hursto75, 2017-06-07:
I will do that.

RolandChen, 2017-09-23:
I've written a Heli 120 CCPM mixer already, and the heli could fly with these code. 
ref: 
[https://github.com/RolandChen/betaflight-Heli_120-NOT_TESTED](url)
[https://www.helifreak.com/showthread.php?t=790976](url)

RolandChen, 2017-09-23:
I hope that Heli120 not to be dropped from BF.  (not BFC)

mikeller, 2017-09-23:
@RolandChen: We are happy to re-add them again, if there is working mixer code for them -  as it was in 3.1.7, the heli mixers were listed, but there was no working code for them. Can you please open a pull request for your mixer implementation?

jflyper, 2017-10-23:
Closed via #4296

closed 2017-10-23 by jflyper