betaflight/betaflight #3042: Possibility to Assign ADC Inputs same way as other resources?

SoulPictures, 2017-05-07:
hi there, 

I'm playing around with the AnyFCF7 is there a possibility to add the ADC Channels to be assignable in the resource list as it is possible with motor channels? Currently I can only select (OnboardADC or ESC) maybe this would also give some flexibility with other boards where it could be possible to use an external current sensor connected to the RSSI channel instead of "hacking and bypassing" the internal Current sensor on boards like the OmnibusProF4SD

best regards arne

jflyper, 2017-05-07:
If speaking of the possibilities, yes, but time frame is unknown.

BTW, iirc, RSSI input of the OMNIBUS F4 Pro is not ADC capable...
(That is why "Software PWM based RSSI (Auto ranging version) #2292" is proposed.)

ledvinap, 2017-05-07:
I did start it some time ago, https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/adc_stm32f30x.c, maybe something can be reused.

digitalentity, 2017-05-07:
I've added possibility to use multiple ADCs - some code might be reused as well https://github.com/iNavFlight/inav/pull/1621

jflyper, 2017-10-23:
It's been worked on #3898.

closed 2017-10-23 by jflyper