betaflight/betaflight #3043: BF3.2 - strange log when doing full roll

DeepnineRC, 2017-05-07:
Hi,
i tested the current master today. See at the end of the log file. I don't know what happend.

Roll axis freaked out. I was doing a hard turn the roll started to oscillate. I disarmed and let it crash.

An body an idea? Error in firmware or error in hardware or error in pilot?

[strange_log.BFL.txt](https://github.com/betaflight/betaflight/files/981718/strange_log.BFL.txt)

ledvinap, 2017-05-07:
Looks like some kind of overflow in setpoint. 
[image]
[image]

DeepnineRC, 2017-05-07:
I had crash_recovery (defaults) on for testing. Maybe there is the problem.

rav-rav, 2017-05-09:
Strange indeed, no idea why the dTerm goes crazy.

Looks like your rcCommand does not get updated during the freakout for ~1.2sek. Maybe your receiver lost connection during the hard maneuver. Could it be bad soldering?

martinbudden, 2017-05-09:
It's possible that a hard roll is interpreted as a crash by the crash recovery code, but the logs don't seem to indicate that.

The spikiness in the gyro roll trace in the second diagram looks very odd, and seems to indicate faulty gyro output. Have you experienced that on other occasions?

rav-rav, 2017-05-09:
Gyro looks fine to me but in crash recovery the accelerometer is used to determine the new setpoint instead of rcCommand?

In the log stick deflection for roll is at the maximum and has a huge influence on the dTerm. I think it should just be ignored when in crash recovery.
https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c#L430

martinbudden, 2017-05-09:
You are absolutely right that the stick deflection should be ignored for calculating the delta for the crash detection - that was a silly oversight on my part. I'll get a fix in towards the end of the week.

You are correct is assuming that the new setpoint is determined by the craft attitude in crash recovery mode.

As for the traces, I think I made a mistake discerning the colors, and confused `PID_sum[pitch]` with `gyro[roll]`.

ledvinap, 2017-05-09:
@martinbudden: Sorry, I should have taken more time preparing the screenshots ... 

Gyro signal is clean, but something is forcing setpoint; possibly by crash recovery.

The spikes on second screenshot are strange and probably enough to cause the oscillation.

ledvinap, 2017-05-09:
@DeepnineRC :  Can you post cli dump, please?

ledvinap, 2017-05-09:
@martinbudden : stick deflection is probably main culprit - crash recovery mode enter and exit conditions may overlap. 
Entering/exiting recovery mode is performed inside axis loop, possibly leading to inconsistent results (other axis may be affected on same or on next loop). Crash recovery may be easily moved to some slower part/outside of 'pidController'

The spikes are caused by switching out of crash recovery for single cycle, ofset is ~2000, stick position dependent. 
Energy in spikes is high enough to cause oscillation - when angle and angle rate is low enough, recovery is exitted for single/half cycle and entered again. The spikes increase angle/angle rate beyond threshold, so recovery stays on untill angle is reduced again.
This repeats as long as stick is held at extreme. Nice property is that angle and angle rate is bounded, so it should be quite safe to test (if ESC can cope with input)

DeepnineRC, 2017-05-09:
No bad soldering. No bad wire. Soldering is part of my daily job!! All solder joints secured with glue.

I have another log when a fast roll was detected as crash. It starts the the roll and then suddendly levels back to horizon. Very irritating. But no shaking.

# version
# Betaflight / KIWIF4V2 3.2.0 May  6 2017 / 21:48:09 (fed63c2)

# name
name 210new

# resources
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE

# mixer
mixer QUADX

[... 482 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 103
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 89
set pitch_srate = 89
set yaw_srate = 90
set tpa_rate = 10
set tpa_breakpoint = 1650

martinbudden, 2017-05-09:
@ledvinap , you analysis makes sense. I'll make some changes based on your comments.

closed 2018-04-21 by DieHertz