betaflight/betaflight #3048: BetaFlight Configurator does not work after BLHeliSuite passthrough

mclei, 2017-05-08:
I use BLHeliSuite.exe with Wine on Debian Stretch. After running it with passthrough mode of BetaFlight it is not possible to connect with BetaFlight Configurator again till computer restart.

But I found easy solution. When I run:
`stty -F /dev/ttyACM0 min 1`
it is again possible to connect with BetaFlight Configurator.

So the problem is, that BetaFlight Configurator is not setting this parameter of serial port and it should be added. I think it should be easy to fix.

mikeller, 2017-05-10:
That's a good find, thank you very much!

This is a betaflight-configurator issue, and not a firmware issue. Can you please reopen it in https://github.com/betaflight/betaflight-configurator ?

mclei, 2017-05-11:
Oh sorry, I did not realize there is another configurator repository.. Created there: https://github.com/betaflight/betaflight-configurator/issues/504
Thanks.

closed 2017-05-11 by mclei