betaflight/betaflight #3049: SDCard Working on F7s?

kc10kevin, 2017-05-08:
@MJ666 @sambas   Quick question...is SDCard working on F7 boards?  I'm having issues getting it working on the FuryF7.  Tried just about everything.   Works perfectly on the FuryF3 and FuryF4.  Used the same components and checked all connections.  Just get the Fatal Error in the configurator.

New FuryF7 design with STM32F722.   Code below...thanks in advance.

#define USE_SPI_DEVICE_3
#define SPI3_NSS_PIN              PA15
#define SPI3_SCK_PIN              PC10
#define SPI3_MISO_PIN           PC11
#define SPI3_MOSI_PIN           PC12

#define USE_SDCARD
#define SDCARD_DETECT_INVERTED

#define SDCARD_DETECT_PIN                              PD2
#define SDCARD_DETECT_EXTI_LINE                    EXTI_Line2
#define SDCARD_DETECT_EXTI_PIN_SOURCE       EXTI_PinSource2
#define SDCARD_DETECT_EXTI_PORT_SOURCE    EXTI_PortSourceGPIOD
#define SDCARD_DETECT_EXTI_IRQn                    EXTI2_IRQn

#define SDCARD_SPI_INSTANCE              SPI3
#define SDCARD_SPI_CS_PIN                   PA15

#define SDCARD_SPI_INITIALIZATION_CLOCK_DIVIDER 256 // 422kHz
// Divide to under 25MHz for normal operation:
#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER 8 // 13.5MHz

#define SDCARD_DMA_CHANNEL_TX                               DMA1_Stream5
#define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG  DMA_FLAG_TCIF1_5
#define SDCARD_DMA_CLK                                               RCC_AHB1Periph_DMA1
#define SDCARD_DMA_CHANNEL                                     DMA_CHANNEL_0

sambas, 2017-05-08:
Tried latest on F745, seems ok but didn't test logging. Just yesterday did flight log in Inav with about same code.
https://www.dropbox.com/s/jev47coy2uxq4v9/Screenshot%202017-05-08%2017.51.12.png?raw=1

kc10kevin, 2017-05-09:
@sambas - Ok, thanks.  I guess its back to the drawing board and try to figure out what is going on.  

Just weird.  It's the exact same layout as a FuryF4 that works perfectly.  Just modified the couple of pins that are different between the F4 and F7.  Pin assignments for the SDCard between the two are identical.    F4 works, F7 doesn't.    

Thanks again.

jflyper, 2017-05-09:
@kc10kevin How about disabling DMA to isolate the problem.

kc10kevin, 2017-05-10:
@jflyper Thanks for the idea.   I did try disabling DMA and still no joy.  

http://i.imgur.com/cDGhUn6.jpg

Flash chip works fine.  I had both on SPI3 just like the FuryF4 redesign (works fine).  Just for giggles, I took the Flash chip off and commented out the Flash code and still not recognizing the SDCard.  I know the Detect is working properly.

Thanks again for the input.  I'm sure its just something simple that I am missing.

kc10kevin, 2017-05-12:
Ok, did some more testing.  I bought a couple different sdcards to try out.  

1 - PNY microSDHC (already had, not sure model)
2 - Team microSDHC (tusDH16guhs03)  slow - 15mb write
3 - Samsung micoSDHC (mb-mg16e)  faster - 60 mb write

All 3 cards work perfectly on the F4.   Initialize and are recognized right away.

On the F7:

PNY - Fatal error no matter what I do.
Samsung and Team cards - cards show "Filesystem Starting..." in the Blackbox tab when starting from a newly formatted card (using SDFormatter) and stays there.  I then take them and insert them into an F4 and they initialize properly and show up nicely in the Blackbox tab.   Take them back to the F7 and they show up properly in the Blackbox tab.   

Not sure why the PNY is showing fatal error constantly and the Team/Samsung cards show "Filesystem Starting...." from a newly formatted card.  

@sambas @MJ666 - would you mind formatting one of your sdcards and see if you have the same behavior as I am seeing with the file system trying to start?   Not sure if you tested that already or just used already initialized sdcards.

Thanks in advance.

MJ666, 2017-05-12:
@kc10kevin I can give this a try when back home.

kc10kevin, 2017-05-13:
@MJ666 Much appreciated!

MJ666, 2017-05-13:
@kc10kevin tested on my AlienFlight F7 dev board and an freshly formatted SD card is properly detected and initialized. Used an Sandisk 8GB card.

kc10kevin, 2017-05-13:
@MJ666 - Thanks for checking that out.   I'm at a loss.  Hours and hours of testing and still can't figure it out.  Oh well.

blckmn, 2017-05-14:
@kc10kevin check the edge settings on the spi. You'll see what I mean in the bus spi Hal init.

kc10kevin, 2017-05-14:
@blckmn - Thanks!  I was looking at the SPI Hal driver earlier.  I'm a novice to this but I'll see if anything is amiss.

Thanks again.

kc10kevin, 2017-05-16:
@sambas @MJ666 @blckmn More testing.....

I went and assembled one of my earlier version of the board with the STM32F745 chip.   Code is current for this version on here  ([Target](https://github.com/betaflight/betaflight/blob/master/src/main/target/FURYF7/target.h)).  Exact same behavior with the 3 cards I have listed above.  So, I decided to get a card that @MJ666 says initialized properly on his board (Sandisk 8GB card, model sdsdqui-008g).  Popped it in the 745 board and it intialized properly!  Sweet!!  Now, I reformatted the sdcard and put it in the 722 board.  Would not initialize.  :-(  

So, I looked at the bus_spi_hal driver and couldn't make heads or tails of the leading edge code, but I did find one thing that fixed the issue on the 722 board (with the Sandisk 8GB):

Line 75 of bus_spi_hal driver:
`{ .dev = SPI3, .nss = IO_TAG(SPI3_NSS_PIN), .sck = IO_TAG(SPI3_SCK_PIN), .miso = IO_TAG(SPI3_MISO_PIN), .mosi = IO_TAG(SPI3_MOSI_PIN), .rcc = RCC_APB1(SPI3), .af = GPIO_AF6_SPI3, .leadingEdge = false, .dmaIrqHandler = DMA1_ST7_HANDLER },`

722 Target DMA stream:
`#define SDCARD_DMA_CHANNEL_TX DMA1_Stream5`

DMA streams do not match.  Changed so they are the same:

`#define SDCARD_DMA_CHANNEL_TX DMA1_Stream7`

The Sandisk 8GB card works great now on the 722 board.   Initializes beautifully.  745 version uses SPI4, so that is probably why it worked.  Am I on the right track here?

The other cards still do not intialize on either FuryF7 version.  To summarize:

1.  All cards initialize (fresh card) on the F4 boards.
2. Sandisk 8GB - Initializes properly in Fury 722 and 745 boards.
3. Team and Samsung 16GB cards do not initialize.  Shows filesystem starting, then goes to fatal error.  If I put them in an F4 to initialize then put them in the F7, the cards show up nicely.  
4.  PNY 16GB card shows fatal error immediately on putting it in the F7.  Even after initializing in the F4, it still shows fatal error in the F7.

Just to emphasize, I am very confident the design is correct.  I have gone through it for hours.  All connections are correct.  Same design as in the FuryF3 and FuryF4 boards.  Just to be safe, I have ordered one of @sambas ANYFCF7 from Readytoflyquads (hope you get some $$$$ from the sale) to test my cards in to see if the problem persists.  I should have it in a couple of days and I'll report results.

kc10kevin, 2017-05-17:
@sambas - Got your board today from readytoflyquads for testing.  I get the exact same behavior on your board as my board on the freshly formatted cards listed above.   Only the Sandisk will initialize in the F7.   All other cards will not.  As mentioned above, all except the PNY card work fine if initialized in the F4 first.

@MJ666 @blckmn @jflyper - I'm going to keep this issue open as I believe there is something amiss in the F7 code.

Can anyone reproduce my initialization results?

mikeller, 2017-05-17:
@kc10kevin: Yes. I am having the same issues ('card being initialised' never ends) with an F722 board. If put the card into an F4 board to get it initialised, it works fine (most of the time) in the F722 board.

kc10kevin, 2017-05-17:
@mikeller - Thanks for the report!   So I'm not going crazy.  lol

mikeller, 2017-05-17:
Maybe we both are. ;-P

digaus, 2017-05-19:
Seem to have same problem :) Will try to put it into F4 first and see if it then works with the Omnibus F7

kc10kevin, 2017-05-19:
@digaus Format the card first with SDFormatter, put in F4 to initialize.  Then try the F7 to see if it is recognized.  

Please post up results and microsd card brand/model.  Thanks!!

digaus, 2017-05-22:
@kc10kevin 
Sadly I do not have another FC with an SDCard slot so I cannot test this. It is a noname 8gb class 10 SDCard from an ebay vendor called memory zone.

Cheredanine, 2017-05-31:
Just to confirm, scan disk 16gb with Nero f7 - no log :(

hydra, 2017-07-09:
@kc10kevin SPI clock speeds...

closed 2018-04-21 by DieHertz