betaflight/betaflight #3050: Omnibus F7 with dual gyro/acc?

digaus, 2017-05-08:
https://store.myairbot.com/flight-controller/omnibus-f7.html

New Omnibus F7 is available with dual gyro/acc...

Is someone already working on it?

digaus, 2017-05-10:
Saw this PR #3058 ... Will it be included in the next release?

mikeller, 2017-05-10:
Yes it will be included.

digaus, 2017-05-10:
When can we expect the next release?

mikeller, 2017-05-10:
We will start working on it in the near future, and a release date will be announced. Stay tuned. :-)

jflyper, 2017-05-15:
The next release will at least include "Selectable gyro" on dual gyro boards.

digaus, 2017-05-15:
Is the OSD supported on the F7?

mikeller, 2017-05-15:
@digaus: Yes: https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF7/target.h#L112-L117

jflyper, 2017-05-15:
Standard for ALL OMNIBUS.

digaus, 2017-05-16:
Ok just received the F7 and noticed that UART3 is not activated? Is there a reason for it? Cause I need the 3 UARTs  (SBUS, Smartport and Smartaudio)

mikeller, 2017-05-16:
According to the target definition it is because the pins are shared with I2C:

https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF7/target.h#L74-L77

For now, you'll have to build your custom binary, enabling UART3 and disabling I2C, in order to use it.

digaus, 2017-05-17:
@mikeller

Thanks, could I assign UART3 via the CLI? Because I do not know how to create a custom binary

mikeller, 2017-05-17:
No, that's not (quite yet) supported. Will probably be in the next release.

Try this in the meantime:

[betaflight_3.2.0_OMNIBUSF7.zip](https://github.com/betaflight/betaflight/files/1006741/betaflight_3.2.0_OMNIBUSF7.zip)

digaus, 2017-05-17:
@mikeller 
Thank you very much! Will test this evening :)

halninekilo, 2017-05-17:
Having a bit of an issue with the OSD, it get's drawn alright, but the input from the cam isn't displayed.
Verified the cam is working (just swapped the FC anyway), also tried manually chosing the correct TV standard - no dice.

That's some kind of hardware issue, isn't it?

Cheers!

digaus, 2017-05-17:
@halninekilo

Will test tomorrow. Currently have the problem that sbus is not Working... soldered to RC and der sbus_inversion = on 

Works correctly on Omnibus F4... any ideas?

halninekilo, 2017-05-17:
Can confirm, SBUS not working on RC

mikeller, 2017-05-17:
@halninekilo: Yes, the camera video signal gets fed through the board as an analogue signal, so no way for the firmware to stuff that up really. Have you checked your camera signal? Is the lens cap off? ;-P

@digaus, @halninekilo: What UART have you connected your SBus to?

digaus, 2017-05-17:
@mikeller 
Soldered to RC input pin... now soldered on RX1 and it works. Checked RC pin with multimeter and it does not seem to be connected at all.

Also SmartPort is not working on UART3 but on UART6 it is working. So there still seems to be a conflict with UART3

mikeller, 2017-05-17:
I don't have a schematic for the board, so I can't tell where the 'RC' pad is meant to be connected to.

Re UART3, do you have a multimeter and feel confident beeping out which pins the UART3 pads are connected to? It's possible that the commented out definitions for UART3 defines the wrong pins.

digaus, 2017-05-17:
@mikeller 
RX is the 4th PIN and TX the 5th from top right
[image]

mikeller, 2017-05-17:
Ha, turns out the pin definitions in the target definition were swapped. Try this:

[betaflight_3.2.0_OMNIBUSF7.zip](https://github.com/betaflight/betaflight/files/1009262/betaflight_3.2.0_OMNIBUSF7.zip)

halninekilo, 2017-05-17:
Went back to the F4 and cam works like a charm. Hope Airbot will send a replacement F7 soon... 

Anyway, thank you for the rapid fixes mikeller!

digaus, 2017-05-18:
@mikeller
Tanks works like a charm now :) will so more tests this evening

mikeller, 2017-05-18:
Good to hear. As soon as #3111 is merged in you will be able to use nightly builds from http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/ to get a version with working UART3.

digaus, 2017-05-18:
Can confirm that everything seems to work! Got SmartAudio, Telemetry, Lua, SBUS, OSD and DSHOT600! Did not fly yet though :)

One thing I noticed is that the ESCs seem to initialise twice on startup

digaus, 2017-05-18:
Ok I noticed some stuff:
- Cannot Select 'Onboard ADC' for Current Sensor and Voltage Sensor
- SDCard is not working 
- LEDs sometimes flicker with a Color Modifier assigned

digaus, 2017-05-18:
Ok Voltage and Current might be due to changed values which do not fit the configurator... set them manually in the CLI and the same to work. But it do not know what the difference between ibata_scale and ibatv_scale is (both for current?). Also I cannot test the motors with the configurator. Think something has changed and they just wont spin up with it.

digaus, 2017-05-18:
Being able to test the motors is kinda critical for me because the 4in1 ESC has no correct layout and some motors are swapped. To know which one I have to be able to spin them up seperately. 
Is there maybe a new configurator which works correctly with 3.2?

mikeller, 2017-05-19:
@digaus: Thanks for testing, and reporting back.

There is an issue with SD card support on F7 boards: #3049 Read this thread to get hints on how you might get it to work.

To get the Motor / Battery tabs to work with 3.2, you will have to get the latest version of the configurator from GitHub.

jflyper, 2017-05-19:
@digaus 
> LEDs sometimes flicker with a Color Modifier assigned

This is a known issue, too, I guess.

F7 LED strip behaves erroneously #3038

halninekilo, 2017-05-19:
Gave this another shot with a different camera - turns out the other had a broken wire... embarassing.

SD-card logging @8khz worked fine with a pre-initialized SD (disabled DMA because I got burned by the F4 with that).

One litte thing: I selected the KISS24A as volt&current sensors (pulled last confi version from github) and I get the the correct voltage, but no current readings. Is this a known issue? (Artifact #1282)

Cheers

mikeller, 2017-05-20:
@halninekilo: No, KISS24A current reporting used to work before the battery code rework. Can you please open a new issue for this? (Also, it is likely not F7 specific, since ESC voltage reporting seems to be working.)

digaus, 2017-05-23:
Trying 32khz results in 100% CPU usuage, even with ACC disabled... is that normal?

mikeller, 2017-05-23:
@digaus: From testing with a board with an F722: Yes. I suspect there is some optimisation that is not yet done in the F7 build.

RipperDrone, 2017-06-04:
Went hovering a bit with Ominbus F7, flies nicely and very smooth - FC bobbins seem to be very effective.

Can confirm teething issues reported by @digaus 

- double ESC boot (ESC, FC, ESC...?)
- BFC cannot edit voltage / current scale factors, but CLI workaround is ok
- RC solder pad seems dead, had to link S.BUS up to RX1
- 2nd hardware button's function not clear
- FrSky S.Port telemetry (UART3), buzzer, 8/8kHz, DShot 600 all working :-)

jflyper, 2017-06-05:
@digaus 
> Thanks, could I assign UART3 via the CLI? Because I do not know how to create a custom binary
Also I2C2 seems to be disabled anyways: https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF7/target.h#L140

UART is now configurable in the master, and I2C is coming. With these enhancements, you will have a choice of assigning UART or I2C to those pins.

digaus, 2017-06-07:
What about the initial restart? Also the SD Card issues? Are the fixes included in the BF3.2 milestone? I have no other FC with SD Card so no way to get it to work.

halninekilo, 2017-06-07:
Just formated and flew a brand new SD Card - works fine (3.2#1282)

RipperDrone, 2017-07-08:
Double Initiative ist fixed in latest builds, great! Now the main F7 constraint is processing speed / CPU load at higher PID frequencies...

halninekilo, 2017-07-08:
Thats great to hear! Sadly, my OmniF7 won't boot with 1436 :(

RipperDrone, 2017-07-08:
Omnibus F7 running on July 5th Nightly. No probs...

halninekilo, 2017-07-08:
Yeah, that's still working - thanks a lot!

digaus, 2017-07-08:
I am running 1281-1c1cebc with no problems :) do not see a reason to update yet

halninekilo, 2017-08-01:
Just flashed the RC and discovered that I am no longet able to select ESC Sensor and IRC Tramp simultaneously for UART 3... pretty sure that worked before, trying it now just rests the ports config to zero.

This leaves me only with UART6 which does not seem to work :( 

Any suggestions? Should I resource map UART2? Where to?

I really don't wont to give up the ESC telemetry or the Tramp support, once you get used to those features.... ^^

Cheers!

jflyper, 2017-08-02:
@halninekilo 
> Just flashed the RC and discovered that I am no longet able to select ESC Sensor and IRC Tramp simultaneously for UART 3... pretty sure that worked before, trying it now just rests the ports config to zero.

It couldn't have been a case (it is impossible to share a port with ESC sensor and Tramp).

You have your Tramp connected to TX3 pad, you were probably running software serial on PB10 for Tramp.

halninekilo, 2017-08-02:
Went to check and you are right, Tramp is soldered to TX6 but doesnt react to input.

jflyper, 2017-08-02:
@halninekilo I think your have a issue to be supported, and this is not an appropriate place for support activity (Betaflight github issues is a place for discussions on making Betaflight software better.) Can you summarize your issues and post it to the OMNIBUS F7 thread in RCG where I can help you?

closed 2017-08-02 by jflyper

halninekilo, 2017-08-02:
Sorry, but I think I disagree... UART6 not working on the Omnibus F7 target in the BF 3.2 RC should fall under the category of "making Betaflight software better" (at least for Omni F7 owners).

I went back to http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1430/ and it's working like it did before.

jflyper, 2017-08-03:
@halninekilo Can you submit a new issue? Your issue is out of context of this thread.

jflyper, 2017-10-10:
@jamborloi 
I2C is not active on TX3/RX3 unless explicitly configured.
For further assistance, please use the RCG thread.
https://www.rcgroups.com/forums/showthread.php?2884546-Omnibus-f7

jamborloi, 2017-10-13:
@jflyper Hi thanks for the reply.
My vtx VT5804 which should work with IRC Tramp on BF peripheral is not working I only suspect i2c is the default. Any way if it is not active then it might be on the vtx problem. I will check with the manufacturer. Thanks

Fly-addiction, 2017-10-27:
HI, my situation is: when conected by usb without power battery double gyro works fine, I see in sensor tab apparently normal, if plug battery having usb mode works fine too and if disconect usb continue working pretty good. The problem is if i conect battery without usb mode nothing happen the sensor complety static.   I did try change all combination possibles 32/16, 16/16, 16/8, etc, etc. Any Ideas?

Otherwise, when try with the integrated gyro works fine.

jflyper, 2017-10-27:
@Fly-addiction 
Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

goldjacket, 2017-10-29:
Fly-addiction Hi man, Im having the EXACT same issue. wtf. Have you figured this out? Ive spent hours trying to fix this. Please Let me know asap if you know.

thanks

Fly-addiction, 2017-10-29:
Nothing for the moment have to deactive 32kh gyro. Hopping somebody could
help. I think is firmware issue.

goldjacket, 2017-10-29:
Yup underground FPV. Just sent me FC firmware. Works on 8k 8k but not 32
yet. Just glad I can go fly

mooiweertje, 2018-03-21:
I addressed this issue to Banggood. They sent me a new one and it doesn't have this voltage regulator problem.
But this one I turned off the moron_threshhold and calibrate on first arm on. Now it arms on 32/16. I'm using one of the first BF 3.4 dev builds with fast Kalman filters.
It flies amazingly well.

SRTDRE76, 2018-07-04:
On review bee32 F7 whats the function of second button beside the boot button. Is it a quick gyro switch?

DieHertz, 2018-07-04:
@SRTDRE76 Betaflight has no such functionality