betaflight/betaflight #3053: Some SBUS FrSky receivers on BEEBRAIN and NAZE don't work since 3.1.0

andrejpodzimek, 2017-05-08:
I've just got [a new version of the Oversky Super-X V2 classic](http://www.gearbest.com/brush-fpv-racer/pp_615695.html), fitted with FrSky. (These used to come with DSM2 receivers.)

I bound it with my Devo 10 (Deviation v5, 4-in-1 multiprotocol module) and got a steady green light on the receiver. However, no inputs were shown in the Betaflight receiver tab; all channels but throttle were frozen at 1500 and throttle was stuck around ~860. After a trial-and-error all-nighter, I found out that downgrading to Betaflight BEEBRAIN 3.0.1 fixes the problem. Summary:
- Betaflight BEEBRAIN 3.0.1: **works and flies**
- Betaflight BEEBRAIN 3.1.0: broken (no RX input)
- Betaflight BEEBRAIN 3.1.7: broken (no RX input)
- Betaflight NAZE 3.1.7: broken (no RX input)
- Cleanflight BEEBRAIN 2.0.2: broken (no RX input)
- Cleanflight NAZE 2.0.2: broken (no RX input)

An extra observation concerning the RX input: I tried to switch the FC to the `SPEK1024` receiver protocol. And it behaved in a way one might expect when trying to run an `SBUS` receiver on a FC configured for `SPEK1024`: Any stick movement produced lots of random and noisy changes and spikes across multiple channels. With sticks in a stable position, channels would settle on constant (yet quite random) values. But there was definitely a direct response from the RX to the control inputs on the TX, which means that the receiver was bound and working. Just the `SBUS` protocol didn't work or was not interpreted correctly and `SPEK1024` obviously wasn't the right protocol.

What I tried, apart from (or orthogonally to) all the versions listed above:
- `sbus_inversion` set to `on` and `off`. This didn't change anything on the broken versions and had to be `on` for the working old version to work. ;-)
- `serial_update_rate_hz` set to a few values other than 100 (in versions that have this new option).
- All the other serial protocols. On non-3.0.1 versions, most of them showed nothing and 3 showed noise (`SPEK1024`, `SPEK2048`, `SRXL`), as described above.

No matter what I tried, only Betaflight 3.0.1 worked. So it seems that the support for this particular FC+RX combination **got broken between 3.0.1 and 3.1.0**.

andrejpodzimek, 2017-05-08:
Another thing I forgot to mention: BLHeli passthrough doesn't work in Betaflight BEEBRAIN 3.1.7. Yet it does work in 3.0.1. (And I haven't tested any of the other version combinations.)

borisbstyle, 2017-05-18:
The F1 boards are unfortunatelly getting less supported nowadays as they lack flash space.
You will see that features can get broken more and more in the future

andrejpodzimek, 2017-05-18:
I don’t understand why the removal of features starts from vital ones, such as SBUS receivers support. (And has that support been actually removed or could it be just a bug?) Useless features (for this type of aircraft) such as BlackBox would be, IMHO, a much better candidate for removal when it comes to coping with flash memory shortage.

Can one build a stripped-down version without useless features but with latest PID algorithms and — the most importantly — with the latest configuration interface protocol? The problem is that sooner or later the Betaflight and Cleanflight consoles will become incompatible with the aircraft and make it hard to tweak without upgrade options.

Is it possible to build e.g. the latest version without Spektrum, BlackBox and LEDs? I’d be more than happy to build it myself from Git if there’s a howto somewhere.

martinbudden, 2017-05-18:
@andrejpodzimek , SBUS support hasn't been removed from NAZE boards. It however, as you point out, seems to be broken for your particular board.

The developers have limited time and support for F1 boards tends to have lower priority.

If you with to build the software yourself there are some instructions here: https://github.com/cleanflight/cleanflight/tree/master/docs/development . The instructions are for Cleanflight, but the process is similar for betaflight.

andrejpodzimek, 2017-05-19:
@martinbudden Thanks for the update. I understand that this board is not a priority, but if SBUS support was not removed between 3.0.1 and 3.1.0, then this must be indeed a recent bug. Perhaps an issue should remain opened for it, even if nobody fixes it, just for reference. Could #2508 or #2840 be possibly related? I thought I'd try to revert recent changes to `src/main/rx/sbus.{c,h}` and try the resulting build to see if any of them could be causing the RX problem, but it looks like all changes that could make a difference between 3.0.1 and 3.1.0 are just refactoring. Are there any other relevant changes I should try to revert? I'm not familiar with this code base, so I have no idea what else can influence FrSky SBUS receivers. Any guesses would be helpful. Bisecting is also an option, but I'm not sure how many re-flashes the board can take.

martinbudden, 2017-05-19:
Yes, I think it's worth re-opening this to do some investigation.

Firstly can you check if the tip release of `master` works?

If that doesn't work, can you do a bisect to find the actual change? Typically flash can be re-written tens of thousands of times, so I don't think the reflashing is something to worry about. I have boards that I have re-flashed many many times.

andrejpodzimek, 2017-06-04:
Sorry for the delay.
I've just done `git bisect start v3.1.0 v3.0.1`. This eventually identified `7ec2031ce51fb71e8cc9e6eb868e630ddf951086` as the culprit. So I think [this change is the problem](https://github.com/betaflight/betaflight/pull/1654/files). The current latest `master` state does **not** work (same symptoms as above). However, after I did `git revert 7ec2031ce51fb71e8cc9e6eb868e630ddf951086`, rebuilt `BEEBRAIN` and reflashed, it just works. So the latest state of `master` minus that one change works for me. I've just been flying it for ~2 packs and everything looks OK. And it somehow feels better than it did with v3.0.1, less shaky or the like.

mikeller, 2017-06-04:
Good finding, @andrejpodzimek! Looks like the comment in the commit is wrong, and the change _does_ have an effect - at least on F1.

Please test with the binaries in #3201 and submit your feedback.

andrejpodzimek, 2017-06-04:
@mikeller, I think your pull request works. I didn't test your binaries, but simply rebuilt `BEEBRAIN` after `git pull git@github.com:mikeller/betaflight.git fix_sbus_for_f1` into `betaflight`'s `master`. Yes, it flies. :+1:

andrejpodzimek, 2017-06-04:
BTW, the BLHeli problem mentioned [in my comment above](https://github.com/betaflight/betaflight/issues/3053#issuecomment-299927679) is caused by b1844eb5f90bd2d8ea5eff8c50499acf5f99843e So it's completely unrelated. I'll file a different issue to look into it. (It would be awesome to have e.g. `BEEBRAIN_WITHOUT_BLHELI` and `BEEBRAIN_WITHOUT_BLACKBOX`, so that all the functionality is covered even though it doesn't fit at once.)

mikeller, 2017-06-04:
@andrejpodzimek: Have a look at the discussion in https://github.com/cleanflight/cleanflight/pull/2812 to get an idea on why and how F1 targets are maintained today.

ledvinap, 2017-06-13:
@andrejpodzimek : Any chance you can trace SBUS data? Saleae logic analyzer / FTDI usb-serial / oscilloscope?

andrejpodzimek, 2017-06-14:
Nope, I don't have such equipment, sadly enough.