betaflight/betaflight #3055: blinking red led

8bitreid, 2017-05-08:
#### Flashing Red Led
I recently updated my Betaflight F3 with the latest from the Firmware Flasher.  Now, I have a quick blinking Red led.  what does this mean? 
#### Strange Aux 9 readings.
Also, my Aux 9 channel reads like there is constant input... is this from a bad setup of Telemetry?  thanks.

jflyper, 2017-05-09:
@8bitreid 
> Also, my Aux 9 channel reads like there is constant input... is this from a bad setup of Telemetry? thanks.

Are you using XM or XM+?

8bitreid, 2017-05-09:
LR1000. FuriousFPV

jflyper, 2017-05-09:
@8bitreid
A quick look at the FFPV's LR1000 product pages tells me

> - S-bus Compatibility w/ No FC Inverter Necessary

Flashing red LED probably means the receiver signal is not detected.
Try cli command "set sbus_inversion = off".

 > - (8) Channels of Output w/ 9th Channel Dedicated to RSSI

So, your aux 9 is RSSI signal. No wonder it always have a fluctuating input.
You can use the channel as input to RSSI, or, just ignore it, or, can hide it using `set max_aux_channels` CLI command.

8bitreid, 2017-05-09:
I will try this. Thanks.

8bitreid, 2017-05-09:
subs_inversion is off.

~~~
# dump
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A00
resource MOTOR 8 A02
resource MOTOR 9 A08
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
[... 420 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
```
 the blinking red led is still happening

8bitreid, 2017-05-09:
I turned off the beeper input that I set on a channel that wasn't reading.  this stopped the flashing.  Now when I arm I get a steady blinking red led. is this the desired behavior?

jflyper, 2017-05-10:
@8bitreid 
I was stupid. If you were getting inputs on AUX9, then the receiver input was (is) working.

Can you describe "blinking" and "steady blinking"?

8bitreid, 2017-05-10:
The blinking is gone, I removed the beeper switch input. No need for it. The blinking was almost like a SOS but now that I restarted after this. The FC powers on with solid red.

jflyper, 2017-05-10:
@8bitreid 
Hey, was the SOS blinking like "_Repeating pattern of five quick flash followed by slow two (or three) flashes_" ?

8bitreid, 2017-05-10:
Yes.

jflyper, 2017-05-10:
@8bitreid If that was the case, then that's a system error.

Depending on numbers of quick flash:

1: Developer
2: Acc/gyro detection failure
3: Acc init failure
4: Acc not supported
5: Invalid EEPROM contents
6: Flash write error
7: Gyro init failure

I suspect you have some problem with acc/gyro chip or have loaded a wrong firmware target.

HaseebNain, 2017-09-07:
@jflyper is the blinking LED information documented anywhere?

DieHertz, 2017-09-07:
For me it's blinking when not armed and with throttle greater than min_check

mikeller, 2017-09-08:
@8bitreid: https://github.com/betaflight/betaflight/blob/28a488b3d2a9117705bb1547a900f5f7ffef4c92/src/main/drivers/system.h#L25-L33

chrisroselli, 2017-09-10:
Having similar problem. When plugged into USB, blinks once, followed by very fast flashes, then blinks once again. Then loops that same sequence.

mikeller, 2017-09-10:
@chrisroselli: Read https://github.com/betaflight/betaflight/issues/3055#issuecomment-300381671

chrisroselli, 2017-09-10:
@mikeller: Ya saw that, just not sure what to do about a system error.

jflyper, 2017-09-10:
@chrisroselli Is it ONCE after the fast flashes?

chrisroselli, 2017-09-10:
@jflyper Yes

jflyper, 2017-09-10:
@chrisroselli Did it suddenly started? Or after flashing some new firmware?

chrisroselli, 2017-09-10:
It suddenly started. It will take a flash, but doesn't fix the issue.

jflyper, 2017-09-10:
@chrisroselli 
I'm sorry, but let me confirm again. Is it "Five or so rapid flashes followed by a single slow flash, then repeating this pattern" ? Can we see, if all possible, a short movie clip of the blinking pattern?

closed 2017-10-23 by jflyper

Droner79, 2018-05-01:
Jflyper I have a constant flashing red light on flight controller f3 and no response from rx in rx tabs any suggestions?Thx in advance.

jflyper, 2018-05-01:
@Droner79 
I suspect you have RXLOSS condition. Bad receiver configuration (hard or soft) or broken receiver.

As that is not a problem of the Betaflight software itself, 
please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

sammyw12, 2018-05-21:
My flight controller red light flashes at a constant speed and won't arm what could this mean

MikeGoodman, 2019-08-07:
So I know I'm about a year late but I'm using an F3 Evo. I'm new to this so I'm not sure what I've done wrong or if something is broken. The F3 Evo blinks red when it's plugged into a USB and when the throttle is just above idle. I haven't managed to get the motors spinning without using a direct USB connection to my computer. 

I also had an issue earlier today where my drone gave full power by itself while plugged into my computer, shredding my USB cable so please could anyone let me know what could have happened as this thing decided to take off by itself.

Freddy0707, 2019-08-18:
Hi I am getting high reading on aux 10 when connecting receiver
tring to get an fli14 + to work with tiny halk, it talks but wont arm

Freddy0707, 2019-08-18:
Mike 
https://blog.dronetrest.com/quadcopter-not-arming-how-to-solve-in-betaflight/

VodecxFPV, 2019-10-27:
Dear Community
I have the same or a very similar case.
4 slow red light blinks followed with 5 fast flashes. Blue LED always on. After a while there is only the blue LED. I have flashed the firmware, but nothing changed. 
Has anyone found out how to fix it?